X1Plus/X1Plus #80: Enable SMB and/or NFS share from X1C panel UI

michaeljsmalley, 2024-03-18:
**Is your feature request related to a problem? Please describe.**
Persistent file storage with native file manager integration on most operating systems is viable via SMB or NFS mounts. For most users, this method of moving files to/from their printer is more accessible than SCPing files over SSH. Given many users put large SD cards in their printers, then end up not using a lot of the space, this could be a nice feature that gives users a place to store anything they want using their printer as a file server. The caveat of course is that SD cards are single point of failure, so this would be at a users own risk. A warning dialog may be in order during the enabling process.

**Describe the solution you'd like**
In the Configuration menu on the front panel, add an option to enable SMB or NFS, and configure a share name (and maybe rudimentary permissions).

**Describe alternatives you've considered**
SCP is the alternative and already works today.

**Software Information**
 - Current X1Plus Version: 1.0

tehmaze, 2024-03-18:
Instead of NFS I would suggest looking at AFP which would offer "native" intergrations with MacOS & friends https://github.com/Netatalk/netatalk

balika011, 2024-03-18:
You can also use ftp

bulior, 2024-03-19:
How to use ftp ? Cant get into it with root and SSH password

3DJupp, 2024-03-28:
Would be great to see third party file transfer tools, built on open standards such as NFS, SMB etc.
It would be a massive feature to connect to your storage and directly print a certain *.3mf, or even STL, but thats complicated, i don't think we should slice the print files on the display unit 😃

bani6809, 2024-04-09:
imo a webui would be a far better place to manage this stuff. trying to manage nfs mounts and smb mounts via the touchscreen OSK would be a nightmare.

Renegade605, 2025-01-10:
I'd love to see an NFS/SMB mount option in the firmware. Not to use the printer as a NAS because (sorry OP, no offence intended) that's a really bad idea. But, I'd like to rsync timelapse videos off the printer to my server for easier viewing and sharing.

rsync doesn't work over ftp, so my current workaround is to periodically dump the entire contents of the printer SD card to a hidden directory on the server and rsync from there to the shared directories.