X1Plus/X1Plus #81: X1Plus startup failed.

dmiart1235123, 2024-03-18:
**Describe the bug**
When fat32 is not the first partition on sdcard it fails to boot into X1Plus. This is fixed by re-formating sd card to have sd card as first partition. 

**To Reproduce**
Steps to reproduce the behavior:
1. Have two partitions on sd card(how RPI image partitions it)
2. Install X1Plus 
3. Everything will install without error
4. On first boot into X1Plus will show console error "Mouting base layer 00.00.28.55.squashfs\n  X1PLUS startup failed. initramfs shell is available on UART at 1500000 band."

**Expected behavior**
Expected to boot into X1Plus

**Screenshots/Photos**

[image]

jwise, 2024-03-18:
Oh yeah this is definitely a bug.  Probably we should just have the installer cry instead?

psiberfunk, 2024-03-18:
Def. feels like we should tell people to format with a fresh SD card in such a situation and bomb out.

dmiart1235123, 2024-03-18:
Yeah, I would add a warning somewhere to tell people to re-format cards. I am sure there will be many, many people who just reuse sd cards from other projects. Mine was previously used in RPI that's why it has two partitions. The crappy part, not all windows versions play nicely with two or more partitions on sd card. So some people might not even realize they have two partitions on sd card. Windows 10 finally has a proper support for that though. It will not detect ext filesystem though.

psiberfunk, 2024-03-18:
> Yeah, I would add a warning somewhere to tell people to re-format cards. I am sure there will be many, many people who just reuse sd cards from other projects. Mine was previously used in RPI that's why it has two partitions. The crappy part, not all windows versions play nicely with two or more partitions on sd card. So some people might not even realize they have two partitions on sd card.(windows might not assign a letter to other partitions) Windows 10 finally has a proper support for that though. However, It will still not detect ext filesystem.

WSL2 actually does, but that's besides the point.. a proper partition nuking should do it .

jomixlaf, 2024-03-18:
I believe I was the first person (documented) running thru this issue yesterday. In my case it was the SDCard already in the Bambu for recording. When it finished install the CFW it failed with this error, I then format the SD card via the Bambu itself, still an issue. I will try to formet it with Mac OS once I'm at home.

CobraXLazer, 2024-03-28:
Same problem here, my stoke went away!

jwise, 2024-03-28:
Your stoke will soon return.

jwise, 2024-03-28:
Hopefully in X1Plus v1.1.

CobraXLazer, 2024-03-28:
Sounds good, I had formated the sd card with logical partition or something, just reformatted with official sdcard formater, nice to have the boot loader option. Just put the firmware on the sd card and restarted the installation from there. Good work

jwise, 2024-04-01:
Fixed by #130.