X1Plus/X1Plus #82: printing.png not shown when printing

jneilliii, 2024-03-18:
**Describe the bug**
I've uploaded a printing.png into the proper path as described by the wiki via sftp and when printing it still shows the default logo. 

**To Reproduce**
Steps to reproduce the behavior:
1. Upload printing.png to /x1plus/printers/<serial_number>/images/printing.png
2. Start print
3. See x1plus logo

**Expected behavior**
To see my custom logo

**Screenshots/Photos**
[image]

**External Links**
Has this bug been discussed elsewhere that you can link to? Have you checked this repository's discussions section? N/A

**Additional Details:**
 - Printer type (X1/X1C/X1E): X1C
 - X1Plus version: 1.0
 - AP firmware version: 00.00.22.85
 - AP hardware revision: 
 - MC firmware version: 00.00.22.37
 - MC hardware revision: 
 - Toolhead firmware version: 00.00.07.12
 - Toolhead hardware revision:
 - Computer vision model: 00.01.02.02

**Is the bug present on OEM firmware?**
Have you tested whether this bug is also present in OEM firmware? N/A
If it is, which version? *ex. 01.07.02.00*
Are any steps to replicate it different?

**Custom modifications**
x1plus

**Additional context**
N/A

jwise, 2024-03-18:
Did you restart your printer after adding the `printing.png`?

Does a `home.png` work correctly?

jneilliii, 2024-03-18:
home.png works as expected. both files were added at the same time and the printer has been power cycled a couple of times since.

jneilliii, 2024-03-18:
here's the file, I just downloaded directly via sftp just to make sure I got the one that it's using (white on transparent). 

[image]

jphannifan, 2024-03-18:
yea this bug has been around since I added this, and I need to take a closer look at where the image displayed on the print progress screen comes from.

The image displayed for print progress has a static link defined here, and as you can see on line 17 of this patch, it checks if there's a key for it in the settings file. If not, it checks if there's an image on the SD card. if not, it defaults to the X1Plus logo.

https://github.com/X1Plus/X1Plus/blob/main/bbl_screen-patch/patches/printerui/qml/main/HomePage.qml.patch

jwise, 2024-03-18:
Oh, I see what's happened. See screen.qml line 76.  We though that `Home2Page` was just for smaller screen printers.  That's not correct: `Home2Page` is for what's used when printing.

So for now you can work around this by putting your printing.png in `/sdcard/x1plus/${serialNo}/images/home.png`.  Branch `jwise/fix-printingpng` should fix this for realsies but I'm going to bed instead of testing it.

closed 2024-03-19 by balika011

jphannifan, 2024-03-30:
Fixed

commit 95c02fe1 (2024-04-01): Load the printing image from the right place.  (Fixes #82)
commit fcce8c9b (2024-04-01): Load the printing image from the right place.  (Fixes #82) (#141)