X1Plus/X1Plus #87: [Feature Request] Enable reading/writing filament info to NFC stickers for non bambu spools

LayerCakeMakes, 2024-03-18:
I am not certain if this is possible, but I'd love to see support to write filament color and type to a standard NFC sticker put on none bambu spools. (I'd love to also store filament print settings through Orca or something but feature parity with bambu spools would be a great start)

I know it won't be possible with the encrypted bambu stickers but just standard NFC should work I hope if this is standard NFC hardware.

balika011, 2024-03-19:
I suspect to do that we would need to flash custom firmware to the AMS as well.

LayerCakeMakes, 2024-03-19:
> I suspect to do that we would need to flash custom firmware to the AMS as well.

Yeah that's what we ended up on the discord as a requirement too.

Also since there is only NFC hardware between slot 1/2 and 3/4 the chip can only be written to on one side of the spool at a time so it would be necessary to prompt the user to flip the spool over at least once, or the writting could be done outside of the ams with a usb reader/writer initially.

LakesideMiners, 2025-01-22:
wait, the AMS has a writer in it?