X1Plus/X1Plus #90: AMS Layout has like spazzy colours

joe32, 2024-03-18:
when looking at the fialment menu for ams or even the external spool the colours are kinda messed up and weird

[image]

balika011, 2024-03-19:
I'm not sure I i can see the issue here. Can you please boot into firmware R and take a picture of the same menu, to see how it's supposed to look like?

designgears, 2024-03-25:
> I'm not sure I i can see the issue here. Can you please boot into firmware R and take a picture of the same menu, to see how it's supposed to look like?

The background behind the icons makes it really confusing to see the filament level when about 2/3 of the filament is used.