X1Plus/X1Plus #91: <BambuHandy app webcam>

AlexVikingos, 2024-03-18:
X1Plus firmware web camera quality is poor and only works from orca and bambu slicer, but not from mobile app in iPhone

dmiart1235123, 2024-03-18:
Mine is the same quality on Android as on orca/bambu. Maybe some sort of wifi connectivity problem on iphone's end?

campmeharder, 2024-03-19:
I have the opposite, works well on handy app, but the bambustudio on linux is choppy and poor image.

jwise, 2024-04-01:
Converting to discussion; WiFi performance is likely improved in #84 and #85, but it's not clear that this is a problem with X1Plus wifi as much as it is just this printer's wifi being kind of flaky.  Can you reproduce this regression:

1) after you boot into stock FW?
2) after installing the brcmfmac test fw?