X1Plus/X1Plus #94: RTSP for chamber camera.

RobinWhitfield, 2024-03-18:
**Is your feature request related to a problem? Please describe.**
Camera is not accessible when not in cloud mode

**Describe the solution you'd like**
Provide chamber camera via RTSP

**Software Information**
 - Current X1Plus Version: 1.0_rc3

**Additional context**
Also a popular request on OEM, and a frustration since the launch of LAN mode.
[image]

jwise, 2024-03-18:
I thought this was supported with LAN Mode Liveview?

RobinWhitfield, 2024-03-18:
I was under the impression it was only a live video connection mode for Bambu Studio rather than local RTSP.

RobinWhitfield, 2024-03-18:
Ahh, so following some discussion in the Discord, I see I'm just rather out of date with what the situation is.

However I'm going to keep this request open as personally for my printer I have zero issues with it being RTSP, and actually being supported by things (who knew VLC still doesn't support RTSPS).

Perhaps the request is now for an option to switch to RTSP in the gui...

balika011, 2024-03-19:
we could add a switch next to the live view to disable secure layer with some super annoying "have to wait 10 sec before able to press okay" warning messages

RobinWhitfield, 2024-03-19:
Sounds reasonable to me

jngmrks, 2025-01-31:
Yes, please put normal RTSP on the printer. I've been waiting for it for ages. I've tried to do it myself, but unfortunately I can't find where or how to start the stream. Ideally, there would simply be a switch between RTSPS and RTSP. In addition, the current RTSPS always crashes, even in the normal firmware, so it's absolutely useless. My hope would be that RTSP would run stably without crashes.