X1Plus/X1Plus #98: Can't Close X1Plus Shield Mode Dialog

ndoty, 2024-03-18:
**Describe the bug**
When choosing the X1Plus Shield option for lan mode it present a dialog to select either Enable shield and restart or Enable LAN mode only.  I do not want to do either as I am in the middle of a print. Please provide a way to close this dialog with out any changes to LAN Mode.

**To Reproduce**
Steps to reproduce the behavior:
1. Settings
2. General
3. Lan Mode Drop Down select Lan + Sheild

**Expected behavior**
Clicking off the dialog close or an X or cancel button is added.

**Screenshots/Photos**
[image]

**External Links**
Has this bug been discussed elsewhere that you can link to? Have you checked this repository's discussions section? 
I checked Discord and GitHub and did not see this reported as of yet. Likely folks actually choose an option and dont need to cancel.

**Additional Details:**
 - Printer type (X1/X1C/X1E): X1C
 - X1Plus version: ? 99.00.00.00
 - AP firmware version: Cant get to printer menus for this or those below
 - AP hardware revision:
 - MC firmware version:
 - MC hardware revision:
 - Toolhead firmware version:
 - Toolhead hardware revision:
 - Computer vision model:

**Is the bug present on OEM firmware?**NO as this is a X1Plus dialog
Have you tested whether this bug is also present in OEM firmware?
If it is, which version? *ex. 01.07.02.00*
Are any steps to replicate it different?

**Custom modifications**
None

**Additional context**
 I was in the middle of a print so the restart part put a hold on this action.  I will enable once the print is over but i cant even get back to the home screen if the printer decided to display a print related message.

ndoty, 2024-03-19:
Update Print finished and Print Completion dialog did display over the Sheild Dialog. However after dismissing the Print Completion dialog the Sheild one was still there. I chose the LAN only button, to avoid a restart, and that did remove the dialog and did keep my Bambu Cloud option enabled. Bambu Handy of course still displayed Printing related messages as needed.

jphannifan, 2024-03-23:
I will add an exit button

jwise, 2024-04-09:
This is going to intercept 1.2, not 1.1, probably.