X1Plus/X1Plus #100: [FR] Add Spool Management Interface, e.g. Spoolman

reipet, 2024-03-19:
Always have Issues running out of Filament during Printing. Currently there is no way to track the consumption status of the filament on the spool. 

**Describe the solution you'd like**
Integration of Spoolman Interface that is a working solution for Klipper. It has the advantage to manage all your Spools for all your Printers using Spoolman Interface to report the consumption to the applet.
Disadvantage, it needs to run on a separate system, e.g. Raspberry Pi

**Describe alternatives you've considered**
Not know alternatives known at this moment

**Software Information**
 - Current X1Plus Version: all

**Additional context**
Potentially it could be combined with reading the RFID Tag and then automatically selecting the Spool in Spoolman. If writing to RFID is possible the remining filament could we written onto the RFID as well.

AndersWJ, 2024-03-19:
+1
Would very much like to see this too.

kylek29, 2024-03-23:
+1 

I'm all for additional functionality like this.

NorthyIE, 2024-03-26:
+1 on this too :)
As per https://github.com/markhaehnel/bambulab/discussions/15 it looks like information about filament usage might be communicated via MQTT