X1Plus/X1Plus #107: Option to wipe SD card during installation process.

RobinWhitfield, 2024-03-19:
**Is your feature request related to a problem? Please describe.**
Currently the printer will fail to boot if the partition setup is not correct (see #81), and completely wiping all partitions off an SD card is not always simple or obvious (for example on Windows if there's a filesystem it doesn't recognise).

**Describe the solution you'd like**
Add an option within the installer to completely wipe the micro SD card before installing X1Plus.

**Software Information**
 - Current X1Plus Installer: 1.0

bani6809, 2024-03-28:
this will have to be done via installer-clientside before the x1p is copied over. i tried to do code to wipe the sdcard printer side but you can't do that as the sdcard is already in use by that time.

3DJupp, 2024-03-28:
But another question would be the disk size:
X1Plus requries **FAT32**, which is limited to **32GB**, at least on most operating systems.
This is not yet included in the [official documentation](https://github.com/X1Plus/X1Plus/wiki/Installation-Guide#getting-started)

> A MicroSD card, **16GB or larger**, FAT32 formatted. High Endurance or Industrial cards are recommended.

jwise, 2024-04-01:
We should now refuse to install with an unsupported partition configuration.  Actually wiping it during the install is kind of tricky.  What do you think about the resulting state of affairs, @RobinWhitfield ?

bani6809, 2024-04-01:
installer-clientside could ssh in and have a look around before even starting.

shitwolfymakes, 2024-09-09:
I'll be implementing this soon