X1Plus/X1Plus #109: Increase Hotend Cap (more then 300C)

Demon8Core, 2024-03-20:
Just a quick idea. A way to increase the hotend max temp to 350C or more to print PPS-CF or similar things. Or maybe (of course with a warning) increase it up to 420 to be able to print PEI9085.

nelsmar, 2024-03-28:
Has anyone verified if the head unit can handle the temperatures? I can probably work something for this as I constantly am sitting right at 300C and was quite excited that this project was finally avaialble. I print 300-360 on a heavily modified prusa constantly. And if I can get a nozzle to support what I need I would be willing to put the work in as I do PETG (non carbon) as high as 300C even.

Demon8Core, 2024-03-28:
Well shit, I just googled the thermistor (NTC 3950 100K) and it seems like that they are only rated for a maximum of 300C. So not only would it require mc hacking, it would also require to change the thermistor or fake the signal. What would be interesting to what type of thermistor the X1E uses because it goes up to 320C.

nelsmar, 2024-03-28:
> Well shit, I just googled the thermistor (NTC 3950 100K) and it seems like that they are only rated for a maximum of 300C. So not only would it require mc hacking, it would also require to change [...]

There is more to this as well. You have to realize how heat transfers through metal. The sensor is the reference at the point of reference. The prusa i started with is almost completely redesigned and when it reads 270-300c it produces an extrusion of 320-360c. So there is more to consider. I cant say much more. But there is more to this overall.

Bravo-Built, 2024-08-04:
What about opening it up to the X1E temps. It maxes out at 320. In the pictures the only part that is really noticeable is that it’s using X1p style plugs for connections. also after reading on the X1E forms, many were/are having issues sourcing the X1E  hotends. This is what one owner had to say on the issue “As best I can tell, and based on my experiences, all of the X1/P1 hot ends work on the X1E. The only differences seem to be the thermal paste and electrical connectors.
For the heater+thermistor+fan, you’ll need to use either an original X1E parts or parts from a P1S.
For the thermal paste, you should either use the paste that comes with the X1E or some third-party high temp paste. I’m currently using Slice Engineering Boron Nitride paste with success.” Given the descriptions in the full conversation it’s hard to get any clear information about thickness or any specific data but given I have heard of people moving the sensor and achieving higher temperatures in that range that was verified at the time with a handheld unit it’s all seemingly possible. In the few pictures I came across the only difference I have found is it has plugs like a p1s and a label saying it’s for the X1E Does anyone with a better knowledge than me see an issue with that temp range in a X1-C? That extra 20°C. is definitely a powerful element of change allowing for PPA-CF/GF, PPS, PPS-CF/GF with the addition of a low budget chamber heater. If there is some difference that I cannot find in the limited information I have found so far concerning the X1E thermistor would it be possible to fit it with our connector plugs? If possible how far would be ok? Because after all Ultem prints at 325/330C.

jwise, 2024-08-04:
The X1C MC will require hacking to go above 300C indicated, no matter what.  Personally this is not a super high priority for me because adding this capability to consumer custom firmware is a massive footgun... there are a lot of ways to make your printer pretend to be a printer that can print at >300C, but not a lot of ways that don't involve a house fire, and I have real concerns about the mechanical architecture of X1C vs X1E to support this.

Bravo-Built, 2024-08-04:
I have looked through the wiki of both machines and that seems to be much of the same even in a lot of the part numbers. Concerning going to the X1E perimeters Could you further explain? Also I’m not very knowledgeable when it comes to software but I was wondering if you might have a little more information about the firmware part as well. Also to be clear I’m not endorsing any of the actions taken to get to higher temp by relocating sensors I was just using as a point

tabiznet, 2024-08-30:
+1 on this. There are some engineering filaments coming down the pipe that need around 320c. It's got me thinking about getting a Prusa again just because high temp hotends are an easy addon but I'd rather save the money and just get an aftermarket high temp hotend for the X1

jphannifan, 2024-08-30:
My problem with this is that the x1 is not calibrated in the factory to print above 300C. If you acknowledge that you may break your device in the process, then I think it's a reasonable goal. I would bring this discussion to the discord especially now that Bambu is selling more 'engineering' filaments

closed 2024-08-30 by psiberfunk