X1Plus/X1Plus #111: fix the LD_PRELOAD chaos in Python

jwise, 2024-03-21:
**Describe the bug**
For some reason, we need to `LD_PRELOAD` libffi into the Python that we build.  I should do some linker hacking or something to make the build of Python that we distribute not have this problem, because this sucks.

**To Reproduce**
Steps to reproduce the behavior:
1. Try to load `ctypes` into a `python3` that you invoke at the command line without any noise.

sbuller, 2024-03-22:
I was going to suggest using `patchelf` to mess with the `rpath`, but none of the elf files in the python archive are calling for `libffi`. I'm guessing that `ctypes` is trying to load `libffi` itself. I see in `lib/python3.10/ctypes/util.py` that `ctypes` is running `ldconfig` to find the paths to libraries to load. I see there that it considers `ldconfig`, `LD_LIBRARY_PATH`, and `gcc`s linker results.

I think `LD_LIBRARY_PATH` is probably a better option than `LD_PRELOAD`. Alternatively you could edit the system's `/etc/ld.so.conf` (ideally a new file in `/etc/so.conf.d/`), and run `ld-config` to regenerate the cache.

jwise, 2024-03-22:
Yeah, I was thinking of something along the lines of patchelf. Indeed, it is the ctypes native library that goes wrong when loading, and I never really debugged to see whether it was because that shared library was loading the wrong libffi, or no libffi at all - I just forced the “right” libffi to get loaded come hell or high water, and then went about my day :)

I could also I guess just write a wrapper shell script for the Python interpreter itself that does what I need. That would be silly.

sbuller, 2024-03-22:
I haven't yet flashed x1plus since I'm hesitant about creating a bambu account, so can I ask if dropping a file in `ld.conf.so.d` and running `ldconfig` is viable?
That said, I've been doing a lot of NixOS stuff lately, and I've been seeing a lot of wrappers, so I don't think that's a bad solution.

jwise, 2024-03-24:
```
[root@BL-P001-sdcard:~]# LD_LIBRARY_PATH=/opt/python/lib/ /opt/python/bin/python3 /opt/gpiokeys.py
Traceback (most recent call last):
  File "/opt/gpiokeys.py", line 5, in <module>
    import dds
  File "/opt/dds.py", line 1, in <module>
    import ctypes as c
  File "/opt/python/lib/python3.10/ctypes/__init__.py", line 8, in <module>
    from _ctypes import Union, Structure, Array
ImportError: /opt/python/lib/python3.10/lib-dynload/_ctypes.cpython-310-arm-linux-gnueabihf.so: undefined symbol: ffi_prep_cif
```

Ugh, apparently that's not quite enough.  Why in the heck does that not get linked against the right thing?

PR #120 (2024-03-25): fix the need to LD_PRELOAD libffi into python
patchelf python and its libraries to load from /opt/python/lib, and also patchelf ctypes to load libffi, and also strip python libraries to save quite a bit of space (fixes #111)
commit 64e4c15a (2024-03-25): patchelf python and its libraries to load from /opt/python/lib, and also patchelf ctypes to load libffi, and also strip python libraries to save quite a bit of space (fixes #111) (#120)
commit c105ecd3 (2024-03-25): patchelf python and its libraries to load from /opt/python/lib, and also patchelf ctypes to load libffi, and also strip python libraries to save quite a bit of space (fixes #111) (#120)
closed 2024-03-25 by jwise