X1Plus/X1Plus #117: remove nfs / nfsd from kernel

bani6809, 2024-03-23:
i can't see us seriously supporting nfs client let alone nfs server, we should remove and get kernel memory back.

nelsmar, 2024-03-28:
If enough peopole want this I can look into. I personally have a use for this and others may. But making a toggle or a different kernel for "effiecency" might be ideal. Can you explain more in detail what and why you are seeing use in this and how the memory issues are holding you back from something so I can further understand?

jwise, 2024-04-01:
yeah, -1; I don't think that the cost of nfsd is even perceptible on this printer.  I took a guess at '200 KB' and then decided to actually look it up, and dang was I close:

```
joshua@anima:kernel-x1plus$ size `find fs/nfs fs/nfsd fs/nfs_common -iname '*.o'` -t | tail -n1
 189505	  1611	  1828	192944	 2f1b0	(TOTALS)
```

(that's text, data, bss, total decimal, total hex)

jwise, 2024-04-09:
closing wnf

closed 2024-04-09 by jwise