X1Plus/X1Plus #121: [feature request] Show remaining printtime on locked screen

tscharly, 2024-03-25:
We locked the printer @ our spaceship and people are complaining that they cannot see the remaining printing time.

jwise, 2024-03-25:
Yeah, good idea.  Adding stuff to the lock screen is easy.  Having a good config UI for it is hard.  Do you have ideas of what in general you'd like to see on the lock screen?

tscharly, 2024-03-26:
-remaining print time
-editable Text
-IP

john-f-chamberlain, 2024-03-26:
- Picture of the model being printed would be cool

tscharly, 2024-03-27:
-actual printing speed

tscharly, 2024-03-28:
Stop button on lockscreen

psiberfunk, 2024-03-28:
> Stop button on lockscreen

Uhm … no ? That defeats the purpose

tscharly, 2024-03-28:
> > Stop button on lockscreen
> 
> Uhm … no ? That defeats the purpose of a lock screen .. which is to prevent people from making alterations to the status of the system

Also prevents people from stoping printer if shit is happening. Maybe opt in/opt out for stop-button. 

Next idea:
- show layer height on lockscreen

psiberfunk, 2024-03-28:
> > > Stop button on lockscreen
> > 
> > 
> > Uhm … no ? That defeats the purpose of a lock screen .. which is to prevent people from making alterations to the status of the system
> 
> Also prevents club members from stoping printer if shit is happening. Maybe opt in/opt out for stop-button.
> 
> Next idea:
> 
>     * show layer height on lockscreen

That's what the power button at the back is for I guess ?  I mean, if you don't want messing with stuff, that's what the lockscreen is for.. if you DO want them to be able to stop things.. then don't have a lock screen set?