X1Plus/X1Plus #123: Allow setting setting chamber temperature

kiler129, 2024-03-25:
**Is your feature request related to a problem? Please describe.**
Users of X1 and P1 series can pre-heat their chamber using the bed. However, this is rather cumbersome. Many decide to install a 3rd-party PTC heater in the chamber. However, as of now there's no way to control it from either the GCode nor the printer screen itself.

**Describe the solution you'd like**
1. It would be ideal if the chamber temperature input could be enabled on X1
2. If possible, the chamber heater should be controllable via standard GCode when slicing, enabling it natively.

**Describe alternatives you've considered**
Alternatively, if not possible to retro-fit/enable chamber heating adding a dedicated UI for a 3rd party heater would be a great alternative.

**Additional context**
Naturally the heating element must be controlled externally, for safety reasons. This means the printer needs to expose only the desired temperature for the heater and current chamber temperature. The later part is already possible via MQTT. While I can see relying on two wifi connections for communicating with an LED controller, the same cannot be said about a heater.

As solutions vary, I think the chamber heater temperature should be simply exposed as I don't think the AP board will be able to control the heater like it does on X1E. If there's no local communication port a heater can use (CAN/UART/etc) I believe wiring a USB => UART interface and using it to communicate this (and possibly other) information would be ideal.

jwise, 2024-03-25:
Gcode is a little tricky.  We haven't reverse engineered the Gcode parser yet.  Wiring up the UI bits to an internal API is very possible, though.  And having a standardized expansion board is definitely something we are considering...

Yaksavin, 2024-03-27:
I've had really good luck pre-heating the chamber by running the Aux fan on Max while the bed is at max temp and fully raised or a tiny bit below. This seems like it may be more doable without getting into the G-Code.

jphannifan, 2024-03-30:
> I've had really good luck pre-heating the chamber by running the Aux fan on Max while the bed is at max temp and fully raised or a tiny bit below. This seems like it may be more doable without getting into the G-Code.

There's a python script that serves as a sample macro in 1.0 of X1Plus that uses this same tactic and keeps the bed heated and aux fan on until the chamber temp reaches a set temperature. It reads the NTC directly, and then uses gcode for turning on/off the bed and fans 
`/opt/python/bin/python3 /usr/etc/macros/preheat.py 45`
This will keep running until the chamber reports 45 C, then it shuts off.

We have more elaborate plans for user macros coming in the next couple of updates. A big part of that will be a Python library that will simplify these types of macros.

reipet, 2024-05-23:
Might be an option to use an Rpi as extension board and just communicates with internal Board? Like 2 MCUs in Klipper.
This would add many more options to control or run outsourced from main MCU (e,g Spoolman, webcam, etc)

jphannifan, 2024-08-30:
I have shared a python script above which makes the requests of this issue possible, and I have an updated version of this script where you can define a 3mf/gcode file to print when the preheat temp is reached. [preheat.py.zip](https://github.com/user-attachments/files/16822066/preheat.py.zip)

Since we do not have access to the x1e software features as far as chamber temp control goes, and the hardware mod is not something we can recommend anyone do, I'm closing this. A chamber heater can be added easily by slapping a PTC to the inside of the printer, but it's a huge fire risk unless you have a certain amount experience with electronics.

closed 2024-08-30 by jphannifan