X1Plus/X1Plus #126: Lock Screen Password Keypad persists after unlocking when printer is lagging

Johnnybyzhang, 2024-03-27:
**Describe the bug**
In case the printer is lagging, when doing the "swipe to unlock", the password keypad will stay on the screen even after the printer unlocks. This will not affect the normal operation of the screen but it does block some elements on the screen. Going to a keypad screen i.e. temperature setting will cause the keypad to disappear while confirming or exiting the temp menu. Doing a full system restart will fix the bug, restarting screen service may fix it, but I have not tested it.

**To Reproduce**
Steps to reproduce the behavior:
1. Setup the screen lock with a password.
2. Lock the printer, either wait for the timeout or hitting the power button once.
3. wake the printer from sleep
4. spam the "swipe to unlock" you will want to start the swipe before the lock keypad shows up.
5. unlock the printer
6. if the but do not appear, repeat Step 2-5, I managed to reproduce with the above steps after 5 retrials.

**Expected behavior**
The bugged keypad should not appear after the printer is unlocked.

**Screenshots/Photos**
I'm unable to setup VNC, so here's a photo of the bug

[image]

**External Links**
Has this bug been discussed elsewhere that you can link to? Have you checked this repository's discussions section?

**Additional Details:**
 - Printer type (X1/X1C/X1E): X1C
 - X1Plus version: 1.0
 - AP firmware version: 00.00.28.55
 - AP hardware revision: AP05
 - MC firmware version: 00.00.22.37
 - MC hardware revision: MC07
 - Toolhead firmware version: 00.00.07.12
 - Toolhead hardware revision: TH09
 - Computer vision model: 00.01.02.02

**Is the bug present on OEM firmware?**
Have you tested whether this bug is also present in OEM firmware?
If it is, which version? *ex. 01.07.02.00*
Are any steps to replicate it different?

This bug is impossible as there is no screen lock function in OEM firmware

**Custom modifications**
Add details of any software or electronic modifications you've made to your printer here.
- One custom homepage image at default location
- Changed region from China to Asia-Pacific
- Hydra-AMS, this should not affect the result but I included it anyway.

**Additional context**
Add any other context about the problem here.

RobinWhitfield, 2024-03-27:
Previously reported on the discord here: https://discord.com/channels/1041781310299062282/1219419959172534457

jphannifan, 2024-03-27:
I have a fix for this and I need to make a new branch for it. A property was not reloading and it caused weird stuff like this to happen.. It was on /jon/gpiokeys-dds/ but I am moving the work to a new branch. Shouldn't take us long

jphannifan, 2024-03-28:
I have this fixed in https://github.com/X1Plus/X1Plus/commit/20c47c9e5d9751741869789030adc32d307326fe but I'm leaving this open for now. we're probably moving this commit to another branch

jwise, 2024-04-07:
Ok, here's how I reproduce this:

1) Start bbl_screen.
2) Swipe to bring up the buttons and don't enter the passcode.
3) Wait for the version_notify to pop under everything.  When this happens the buttons will disappear (hmm...!).
4) Swipe to bring up the buttons, enter the passcode.
5) Dismiss the version_notify that is on the home screen.
6) Lock the screen again (Ctrl+O).
7) Swipe to unlock.  Now you're toast.

I'll figure it out from here.

PR #157 (2024-04-07): Avoid leaving bad NumberPads around on the DialogStack if the lock screen is interrupted
Previously, we used this sort of hokey `ersatzDialogStack` and tried to mirror the real `dialogStack`'s state onto it (because the real `dialogStack` was hidden beneath the lock screen).  This was fine except if another dialog came in the middle, which would result in the `NumberPad` becoming very upset, and maybe-or-maybe-not popping the right thing off the stack, and leaving the system in an inconsistent state.

I don't know why I didn't do this to begin with, but this time I just shadow the `dialogStack [... cut]
commit 68ca195c (2024-04-07): Avoid leaving bad NumberPads around on the DialogStack if the lock screen is interrupted (#157)

Previously, we used this sort of hokey ersatzDialogStack and tried to mirror the real dialogStack's state onto it (because the real dialogStack was hidden beneath the lock screen). This was fine except if another dialog came in the middle, which would result in the NumberPad becoming very upset, and maybe-or-maybe-not popping the right thing off the stack, and leaving the system in an inconsistent state.

I don't know why I didn't do this to begin with, but this time I just shadow the dialogStack s [... cut]
closed 2024-04-07 by jwise