X1Plus/X1Plus #128: With no AruCo code and detection on, Z-offset is wrong.

RobinWhitfield, 2024-03-27:
**Describe the bug**
When you have build plate detection enabled but aren't using a build plate with an AruCo code, after confirming the detection error the Z-offset for printing is too high.

**To Reproduce**
Steps to reproduce the behavior:
1. Ensure build plate detection is switched on
2. Have a build plate installed with the AruCo code covered up, or one with no AruCo code.
3. Accept the error about no plate being detected
4. Watch as your first layer is put down too high.

**Expected behavior**
Standard levelling and mesh correction with zero z-offset, resulting in a correctly laid down first layer

**External Links**
[https://github.com/bambulab/BambuStudio/issues/3329](https://github.com/bambulab/BambuStudio/issues/3329)

**Additional Details:**
 - Printer type (X1/X1C/X1E): X1C
 - X1Plus version: 1.0

**Is the bug present on OEM firmware?**
Have you tested whether this bug is also present in OEM firmware?
Yes - Latest 01.07.03.00

**Additional context**
Reported to us by *~ bitbash* on the Discord. As much as this is an OEM bug, if Bambu don't fix it in the upcoming beta, we really should look to fix this ourselves.

RobinWhitfield, 2024-03-27:
**Issue on hold awaiting testing on Bambu's latest beta, unless someone has time to burn to look at fixing it before then**

jphannifan, 2024-03-29:
You can view the z offset that the printer is using by searching syslog for "z_trim". Here are the results of a failed buildplate detection. My plate is set to textured and I had a piece of tape covering the marker. It failed, gave me a warning, and after I continued the print, it set the correct z offset value. -0.04 is the default for textured PEI, so I would say this is functioning correctly and these issues are not related to z offset 
[image]

Flyrian, 2024-05-16:
can confirm the issue on 01.07.02.00, 01.07.03.00 and 01.08.00.00

jphannifan, 2024-05-16:
> can confirm the issue on 01.07.02.00, 01.07.03.00 and 01.08.00.00

You are confirming the issue on OEM or on X1Plus or both?

To help troubleshoot issues like these, have had a PR approved a few weeks back that adds build plate ID and z offset logging to X1Plus. Every time you run a print, your plate ID and z offset are logged to /tmp/x1plus_data.log

Could you check this log file? I would like to know the following:
- the expected plate ID and offset vs. the actual ID and offset
- is there a way to trigger this bug aside from running a print with build plate detection enabled, while using a build plate without an aruco code?

Flyrian, 2024-05-17:
sorry, I'm OEM.