X1Plus/X1Plus #131: [bug report] Bed rammed into nozzle

tscharly, 2024-03-28:
Few days ago i had to pause a print job because of a blob of filament around the nozzle which i had to remove.
Manually moved the bed down and the printhead to the backside.
After resuming printjob, the bed was hammered into the nozzle at full speed and force at two different positions.

**Additional Details:**
 - Printer type: X1C
 - X1Plus version: 1.0

tscharly, 2024-03-28:
[image]

[image]

RobinWhitfield, 2024-03-28:
I have a suspicion that this is an OEM bug.

When you say you moved the bed down to clear the blob of filament, did you move it back up *at all* after clearing it?

Can you remember *exactly* what movements you requested from the printer before and after clearing the clog?

Moving the bed really should not be possible at all when you've paused a print as there's no way for the printer to home Z again without potentially colliding with a print. This option should be disabled.

scooterthedev, 2024-03-29:
Funny enough... I just had the same thing happen to me today, but not while printing. I went to the console, and clicked on lower bed, and the bed flew up, and with the nozzle with so much power, I actually jumped lol. Now there is a huge hole, like the same size as in the photo above, in my bed :(.

RobinWhitfield, 2024-03-29:
@scooterthedev, can you clarify exactly what you did and what you were doing immediately beforehand?

Your comment mentions the console, but I'm guessing you don't mean the G-code/SSH console built into X1Plus.

Which x1plus version are you running?

jphannifan, 2024-03-30:
> Funny enough... I just had the same thing happen to me today, but not while printing. I went to the console, and clicked on lower bed, and the bed flew up, and with the nozzle with so much power, I [...]

Which button did you press? 

Some extra precaution must be taken when using the console. 

From your description, it sounds like you entered the gcode command generated from pressing the button on the console screen called "Bed Down" or "Bed Up"? If left unedited, this sends the command
`G0 Z10 F1200`

There are no hard endstops on this printer, so I would advise using the lowest acceleration values and smallest increments for movement when testing any gcode commands for motion control.

jphannifan, 2024-03-30:
The problem @tscharly described is a different issue, because that is not gcode that the user has intentionally sent to the device. 

I have had a very similar bug on my device, and I found that it occurs both on stock firmware and X1Plus. It seems to be triggered when a print is paused and resumed, or when a print is canceled and the "Print Again" button is pushed. When the bug occurs, it abruptly increases toolhead acceleration and can cause the steppers to stall or the bed/toolhead to crash into walls. 

I broke my toolhead cover trying to troubleshoot this, so unfortunately I don't have much useful info

scooterthedev, 2024-03-31:
> @scooterthedev, can you clarify exactly what you did and what you were doing immediately beforehand?
> 
> Your comment mentions the console, but I'm guessing you don't mean the G-code/SSH console built into X1Plus.
> 
Yes I meant the G-Code console built into X1Plus. I just selected the bed to go down within the console buttons.

> Which x1plus version are you running?
>
I am using V1.0.

jphannifan, 2024-03-31:
> Yes I meant the G-Code console built into X1Plus. I just selected the bed to go down within the console buttons.

The Gcode console doesn't do any interpretation, and there are no callbacks, so it just checks to make sure the printer is idle before executing a new command. 

Also, the coordinates of the toolhead are not exposed to the console, so if you tell it to move G0 X200 (relatively) and the toolhead is not homed, the toolhead will potentially damage itself unless software endstops have been defined ahead of time. 

I think we're going to add a clear indicator to the UI that will tell us whether the toolhead is homed or not. Then disable any relative motion commands if it's not homed.

jwise, 2024-04-01:
This is also presumably possible if the printer happened to be in absolute mode when a bed lower command was given, right?

scooterthedev, 2024-04-01:
> This is also presumably possible if the printer happened to be in absolute mode when a bed lower command was given, right?

What do you mean by absolute mode? Do you mean when the printer hasn't been homed yet?

jwise, 2024-04-01:
"Move bed down" apparently emits a `G0 Z-10` command, and "Move bed up" apparently emits a `G0 Z10` command.  If the printer has not first received a `G91` command, then "move bed up" will not ask the printer to "move the bed up by +10mm", but instead "move the bed up to the absolute position +10mm", which happens to be where the nozzle lives -- bang!

So probably the 'move bed down / move bed up' commands need to be prefaced with `G91\n`.

Ugh, sorry about this, @scooterthedev.  I'm not sure what's up with @tscharly 's case (I still don't understand how we could have changed anything that would cause that, and I also wonder if this is an OEM bug), but at least for the Console page, f815e19 should fix this.