X1Plus/X1Plus #137: Feature Request for X1 Plus Firmware: AMS Filament Path Purge Function

billy12, 2024-03-30:
Is your feature request related to a problem? Please describe.
I frequently encounter an issue with the Automatic Material System (AMS) on the X1 Plus where the filament becomes brittle and breaks, leaving pieces inside the AMS. Currently, the AMS enters a loop of trying to retract the broken filament, with no option to extrude or push it out. This leads to the necessity of manually disassembling the AMS to remove the broken piece, which is time-consuming and disrupts the printing process. The AMS fails to recognize that the filament has snapped and continues to attempt retraction, even when pressing 'retry'.

Describe the solution you'd like
I propose the implementation of a manual control feature for each filament path in the AMS. This feature would allow users to directly command the AMS to either retract or extrude filament from a specific path. For instance, in the event of a filament breakage within the AMS, the user could manually instruct the AMS to extrude from the affected path (e.g., the fourth path). This capability would enable the user to dislodge and remove broken filament pieces more effectively, especially when combined with the removal of the PTFE tube to access the stuck piece. This manual control feature would provide a more hands-on approach to resolving filament breakages and obstructions, giving users greater flexibility and control over the filament handling process.

Describe alternatives you've considered
As a workaround for the issue of filament breakage in the AMS, I have attempted manually feeding filament through the PTFE tube at the back of the AMS while it is in retraction mode. This method can occasionally help in guiding the filament through the system but often leaves broken pieces inside the AMS, leading to further complications. Additionally, this technique does not provide a solution for filament pieces that break just at the tip of the extruder. Although removing these pieces is somewhat easier, it still requires manual intervention and disrupts the printing process. This workaround, while occasionally effective, does not offer a comprehensive solution to the problem and highlights the need for more direct control over the filament paths in the AMS.

Software Information
 - Current X1Plus Version: X1P 1.0.0

Additional context
Adding manual control for each filament path would greatly enhance the functionality of the AMS, making it more adaptable to various scenarios that automated systems may not adequately address. It would empower users to respond more effectively to unexpected issues like filament breakages, ultimately improving the overall user experience and reducing downtime.

jphannifan, 2024-03-30:
So basically like an extrude button for each tray of the AMS? That's a good idea. We have a new macro feature in the works and I think we'll use this idea.

jphannifan, 2024-04-01:
> we would probably need CFW on the AMS for this; I don't think AP CFW can handle this (at least, unless we know of a secret command to the AMS over bambu bus that we can emit from `forward`)

we can still do it with gcode though can't we? we have the mapping of the AMS available, and we can get and set the active tray with PrintManager.feeders.

jwise, 2024-04-01:
I thought the AMS's filament handling motions were all in hard macros.  M620 and T are hard macros that implement almost all of that, right?  We don't get any individual control over the motors in the AMS from Gcode, I thought.

jphannifan, 2024-04-01:
> I thought the AMS's filament handling motions were all in hard macros. M620 and T are hard macros that implement almost all of that, right? We don't get any individual control over the motors in the AMS from Gcode, I thought.

I still need to test this out, but this person's documentation of AMS-specific gcode is what I was basing my initial comment here on. https://github.com/ansonl/topo-map-post-processing/blob/master/Bambu-gcodes.md

As long as we keep track of AMS properties in QML, we should be able to extrude or "respool" each tray on command. That's the idea anyway