X1Plus/X1Plus #153: /userdata/log/recorder is not being cleaned out

bani6809, 2024-04-04:
it seems /userdata/log/recorder is not being cleaned out and can fill up to 100%

we should either warn about this, or automatically prune old data.

i'm pretty sure there is little reason to retain all these forever:

```
[root@BL-P001-sdcard:/userdata/log/recorder]# ls -la | more
drwxr-xr-x    3 root     root         32768 Apr  4 05:00 .
drwxrwxrwx    8 root     root         10240 Apr  3 09:40 ..
-rw-r--r--    1 root     root       1363603 Mar 29 03:30 CMN_2024-03-28_19-15-22.41.bin.gz
-rw-r--r--    1 root     root       1369757 Mar 29 03:45 CMN_2024-03-28_19-30-36.42.bin.gz
-rw-r--r--    1 root     root       1446253 Mar 29 04:01 CMN_2024-03-28_19-45-49.43.bin.gz
-rw-r--r--    1 root     root       1569432 Mar 29 04:16 CMN_2024-03-28_20-01-02.44.bin.gz
-rw-r--r--    1 root     root       1412165 Mar 29 04:31 CMN_2024-03-28_20-16-19.45.bin.gz
-rw-r--r--    1 root     root       1331668 Mar 29 04:46 CMN_2024-03-28_20-31-32.46.bin.gz
-rw-r--r--    1 root     root       1328524 Mar 29 05:02 CMN_2024-03-28_20-46-45.47.bin.gz
-rw-r--r--    1 root     root       1314262 Mar 29 05:17 CMN_2024-03-28_21-01-59.48.bin.gz
-rw-r--r--    1 root     root       1314800 Mar 29 05:32 CMN_2024-03-28_21-17-12.49.bin.gz
-rw-r--r--    1 root     root       1314093 Mar 29 05:47 CMN_2024-03-28_21-32-25.50.bin.gz
-rw-r--r--    1 root     root       1311940 Mar 29 06:02 CMN_2024-03-28_21-47-38.51.bin.gz
-rw-r--r--    1 root     root       1313150 Mar 29 06:18 CMN_2024-03-28_22-02-51.52.bin.gz
-rw-r--r--    1 root     root       1314978 Mar 29 06:33 CMN_2024-03-28_22-18-05.53.bin.gz
-rw-r--r--    1 root     root       1311653 Mar 29 06:48 CMN_2024-03-28_22-33-18.54.bin.gz
-rw-r--r--    1 root     root       1314356 Mar 29 07:03 CMN_2024-03-28_22-48-31.55.bin.gz
-rw-r--r--    1 root     root       1316758 Mar 29 07:18 CMN_2024-03-28_23-03-44.56.bin.gz
-rw-r--r--    1 root     root       1315815 Mar 29 07:34 CMN_2024-03-28_23-18-57.57.bin.gz
-rw-r--r--    1 root     root       1310930 Mar 29 07:49 CMN_2024-03-28_23-34-11.58.bin.gz
-rw-r--r--    1 root     root       1313442 Mar 29 08:04 CMN_2024-03-28_23-49-24.59.bin.gz
-rw-r--r--    1 root     root       1316017 Mar 29 08:19 CMN_2024-03-29_00-04-37.60.bin.gz
-rw-r--r--    1 root     root       1312889 Mar 29 08:35 CMN_2024-03-29_00-19-50.61.bin.gz
-rw-r--r--    1 root     root       1313254 Mar 29 08:50 CMN_2024-03-29_00-35-03.62.bin.gz
-rw-r--r--    1 root     root       1313380 Mar 29 09:05 CMN_2024-03-29_00-50-17.63.bin.gz
-rw-r--r--    1 root     root       1316093 Mar 29 09:20 CMN_2024-03-29_01-05-30.00.bin.gz
-rw-r--r--    1 root     root       1313303 Mar 29 09:35 CMN_2024-03-29_01-20-43.01.bin.gz
-rw-r--r--    1 root     root       1315796 Mar 29 09:51 CMN_2024-03-29_01-35-56.02.bin.gz
-rw-r--r--    1 root     root       1317106 Mar 29 10:06 CMN_2024-03-29_01-51-09.03.bin.gz
-rw-r--r--    1 root     root       1315864 Mar 29 10:21 CMN_2024-03-29_02-06-22.04.bin.gz
-rw-r--r--    1 root     root       1312493 Mar 29 10:36 CMN_2024-03-29_02-21-36.05.bin.gz
-rw-r--r--    1 root     root       1312365 Mar 29 10:52 CMN_2024-03-29_02-36-49.06.bin.gz
-rw-r--r--    1 root     root       1315493 Mar 29 11:07 CMN_2024-03-29_02-52-02.07.bin.gz
-rw-r--r--    1 root     root       1310958 Mar 29 11:22 CMN_2024-03-29_03-07-15.08.bin.gz
-rw-r--r--    1 root     root       1312790 Mar 29 11:37 CMN_2024-03-29_03-22-28.09.bin.gz
-rw-r--r--    1 root     root       1315018 Mar 29 11:52 CMN_2024-03-29_03-37-42.10.bin.gz
-rw-r--r--    1 root     root       1318163 Mar 29 12:08 CMN_2024-03-29_03-52-55.11.bin.gz
-rw-r--r--    1 root     root       1314255 Mar 29 12:23 CMN_2024-03-29_04-08-08.12.bin.gz
-rw-r--r--    1 root     root       1317558 Mar 29 12:38 CMN_2024-03-29_04-23-21.13.bin.gz
-rw-r--r--    1 root     root       1318528 Mar 29 12:53 CMN_2024-03-29_04-38-34.14.bin.gz
-rw-r--r--    1 root     root       1313231 Mar 29 13:09 CMN_2024-03-29_04-53-48.15.bin.gz
-rw-r--r--    1 root     root       1313148 Mar 29 13:24 CMN_2024-03-29_05-09-01.16.bin.gz
-rw-r--r--    1 root     root       1315981 Mar 29 13:39 CMN_2024-03-29_05-24-14.17.bin.gz
-rw-r--r--    1 root     root       1315884 Mar 29 13:54 CMN_2024-03-29_05-39-27.18.bin.gz
-rw-r--r--    1 root     root       1312766 Mar 29 14:09 CMN_2024-03-29_05-54-40.19.bin.gz
-rw-r--r--    1 root     root       1315415 Mar 29 14:25 CMN_2024-03-29_06-09-54.20.bin.gz
-rw-r--r--    1 root     root       1316711 Mar 29 14:40 CMN_2024-03-29_06-25-07.21.bin.gz
-rw-r--r--    1 root     root       1315116 Mar 29 14:55 CMN_2024-03-29_06-40-20.22.bin.gz
-rw-r--r--    1 root     root       1314286 Mar 29 15:10 CMN_2024-03-29_06-55-33.23.bin.gz
-rw-r--r--    1 root     root       1315819 Mar 29 15:26 CMN_2024-03-29_07-10-46.24.bin.gz
-rw-r--r--    1 root     root       1316905 Mar 29 15:41 CMN_2024-03-29_07-26-00.25.bin.gz
-rw-r--r--    1 root     root       1313418 Mar 29 15:56 CMN_2024-03-29_07-41-13.26.bin.gz
-rw-r--r--    1 root     root       1314630 Mar 29 16:11 CMN_2024-03-29_07-56-26.27.bin.gz
-rw-r--r--    1 root     root       1316991 Mar 29 16:26 CMN_2024-03-29_08-11-39.28.bin.gz
-rw-r--r--    1 root     root       1315908 Mar 29 16:42 CMN_2024-03-29_08-26-52.29.bin.gz
-rw-r--r--    1 root     root       1311877 Mar 29 16:57 CMN_2024-03-29_08-42-06.30.bin.gz
-rw-r--r--    1 root     root       1315202 Mar 29 17:12 CMN_2024-03-29_08-57-19.31.bin.gz
-rw-r--r--    1 root     root       1316101 Mar 29 17:27 CMN_2024-03-29_09-12-32.32.bin.gz
-rw-r--r--    1 root     root       1314319 Mar 29 17:43 CMN_2024-03-29_09-27-45.33.bin.gz
-rw-r--r--    1 root     root       1314301 Mar 29 17:58 CMN_2024-03-29_09-42-58.34.bin.gz
-rw-r--r--    1 root     root       1316803 Mar 29 18:13 CMN_2024-03-29_09-58-12.35.bin.gz
-rw-r--r--    1 root     root       1318512 Mar 29 18:28 CMN_2024-03-29_10-13-25.36.bin.gz
[... 215 lines cut ...]
-rw-r--r--    1 root     root       1748673 Apr  3 06:58 CMN_2024-04-02_22-43-10.60.bin.gz
-rw-r--r--    1 root     root       1784785 Apr  3 07:13 CMN_2024-04-02_22-58-28.61.bin.gz
-rw-r--r--    1 root     root       1784771 Apr  3 07:29 CMN_2024-04-02_23-13-46.62.bin.gz
-rw-r--r--    1 root     root       1781908 Apr  3 07:44 CMN_2024-04-02_23-29-03.63.bin.gz
-rw-r--r--    1 root     root       1631558 Apr  3 07:59 CMN_2024-04-02_23-44-22.00.bin.gz
-rw-r--r--    1 root     root       1482143 Apr  3 08:14 CMN_2024-04-02_23-59-37.01.bin.gz
-rw-r--r--    1 root     root       1406807 Apr  3 08:30 CMN_2024-04-03_00-14-50.02.bin.gz
-rw-r--r--    1 root     root       1409626 Apr  3 08:45 CMN_2024-04-03_00-30-03.03.bin.gz
-rw-r--r--    1 root     root       1402459 Apr  3 09:00 CMN_2024-04-03_00-45-16.04.bin.gz
-rw-r--r--    1 root     root       1401905 Apr  3 09:15 CMN_2024-04-03_01-00-29.05.bin.gz
-rw-r--r--    1 root     root       1399968 Apr  3 09:30 CMN_2024-04-03_01-15-43.06.bin.gz
-rw-r--r--    1 root     root       1154536 Apr  3 09:40 CMN_2024-04-03_01-30-56.07.bin.gz
-rw-r--r--    1 root     root       1609156 Apr  4 02:57 CMN_2024-04-03_01-40-23.08.bin.gz
-rw-r--r--    1 root     root       1797582 Apr  4 03:13 CMN_2024-04-03_18-57-47.09.bin.gz
-rw-r--r--    1 root     root       1799707 Apr  4 03:28 CMN_2024-04-03_19-13-05.10.bin.gz
-rw-r--r--    1 root     root       1801270 Apr  4 03:43 CMN_2024-04-03_19-28-24.11.bin.gz
-rw-r--r--    1 root     root       1794230 Apr  4 03:59 CMN_2024-04-03_19-43-44.12.bin.gz
-rw-r--r--    1 root     root       1795990 Apr  4 04:14 CMN_2024-04-03_19-59-03.13.bin.gz
-rw-r--r--    1 root     root       1641510 Apr  4 04:29 CMN_2024-04-03_20-14-22.14.bin.gz
-rw-r--r--    1 root     root       1436559 Apr  4 04:44 CMN_2024-04-03_20-29-38.15.bin.gz
-rw-r--r--    1 root     root       1439727 Apr  4 05:00 CMN_2024-04-03_20-44-51.16.bin.gz
-rw-r--r--    1 root     root       5926278 Apr  4 05:14 CMN_2024-04-03_21-00-05.17.bin
-rw-r--r--    1 root     root             4 Apr  4 05:00 index
drwxr-xr-x    2 root     root          1024 Apr  3 09:40 uconf
```

jwise, 2024-04-09:
I think this happens only if bbl_storage quits and the nanny does not restart it.

riptidewave93, 2024-04-09:
FWIW I am also seeing a ton of these in my `/userdata/log/recorder` dir, even after multiple reboots/prints, so I wonder if BBL changed the retention, or if something is wrong with bbl_storage?

based on `/usr/etc/system/storage.json` it seems the CMN files are whitelisted from getting cleaned maybe?
```
        { "path": "/userdata/log/recorder", "quotaKb": 409600, "persistance": 1, "priority": 10,
          "method": "00020000", "whitelist": "CMN,index,uconf" },
```

jwise, 2024-04-09:
Is your folder more than 400MB?

riptidewave93, 2024-04-10:
> Is your folder more than 400MB?

Close but no, I can keep monitoring it, but ya this may not be a bug and even if it is; it's a BBL one.
```
[root@BL-P001-sdcard:/userdata/log]# du -h ./recorder
2.0K	./recorder/uconf
376.6M	./recorder
```