X1Plus/X1Plus #154: Screen Goes Black, Print Pauses, and SSH not working

scooterthedev, 2024-04-04:
**Describe the bug**
While I was printing for a few hours, in the middle of the print, it suddenly paused in the middle on the infill, and screen shut off. I tried to ssh into the machine, but it wasn't working. All the lights were still on, and the fan was running, but it looks like the printer totally froze. The only way I could get the screen to work, and the printer to respond was to power cycle the machine.

**To Reproduce**
Steps to reproduce the behavior:
Not sure

**Additional Details:**
 - Printer type X1E
 - X1Plus version: 1.0
 - AP firmware version:  00.00.28.55
 - AP hardware revision: AP05
 - MC firmware version: 00.00.22.37
 - MC hardware revision: MC07 
 - Toolhead firmware version: 00.00.07.12
 - Toolhead hardware revision: TH07
 - Computer vision model: 00.01.02.02

**Is the bug present on OEM firmware?**
Have you tested whether this bug is also present in OEM firmware? Not sure, never seen it happen before.

**Custom modifications**
N/A

**Additional context**
I tried tapping the screen to see if it was in sleep mode, but it wasn't. There was no power outage, it just did it randomly. When I power cycled the printer, sadly it didn't continue the print, and it stopped it.

jphannifan, 2024-04-04:
If you haven't already, could you turn on SD logging? 

To enable it, replace {serial} with your printer's serial number and touch this file:
`touch /mnt/sdcard/x1plus/printers/{serial}/logsd`

Then you can either reboot or run
```
/etc/init.d/S01logging restart
/etc/init.d/S93syslog_shim restart
```

And syslog will now save to /mnt/sdcard/log/. That way if you experience this bug again, you will have log output that may explain the memory leak

jwise, 2024-04-04:
Additionally, have you run any other software on the printer or changed any other configurations since the time that you installed X1Plus?  I know that you have said in Discord that you have been experimenting with running other apps on X1Plus.

scooterthedev, 2024-04-04:
> If you haven't already, could you turn on SD logging?
>
> To enable it, replace {serial} with your printer's serial number and touch this file:
>
> ```
> touch /mnt/sdcard/x1plus/printers/{serial}/logsd [...]

Thanks for the explanation. I enabled it just in case it happens again. To answer jwise's question, the only other software I have running is ngrok to tcp forward the ssh connection. It uses extremely extremely low memory (tested with top), and uses barely any cpu either. This shouldn't of affected the printer. Besides for that, everything else is stock.

bani6809, 2024-04-10:
1.1rc1 has perf logging option as well to try to catch this.