X1Plus/X1Plus #162: X1Plus firmware installation doesn't recognize any Bambu X1C printers on network.

lilacmountain, 2024-04-08:
I followed the installation instructions given in the video from Teaching Tech. I rolled back the Bambu firmware successfully (I never installed the firmware upgrade to lock out this firmware) and unlocked the printer. I wrote down the two passwords described. I should now be able to run the X1Plus custom firmware installation software. It will run but does not give me any printers under "Choose a Printer". However, I can see my printer through Orcaslicer or Bambuslicer. I do not have these slicers running while trying to install the new firmware. Is there any advice?

bani6809, 2024-04-09:
are you running your printer in cloud mode or lan mode?

RobinWhitfield, 2024-04-09:
@lilacmountain, if you're on Discord then that would be an easier place to facilitate support: https://discord.gg/4SfCufp6SB

It's unlikely (although certainly possible) that this is actually a bug with the installer.

riptidewave93, 2024-04-09:
We have seen this a few times for users, and a few of the common causes:
* Windows Firewall was blocking the installer
* Printer is on another VLAN/Network
* Slicer was open in the background when running the X1Plus installer

It's worth noting the installer was updated to allow for manual IP entry, so once we get 1.1rc/1.1 out, a manual method will be available to enter your printer's information.

riptidewave93, 2024-04-16:
Closing this out due to no response in over a week, @lilacmountain please come to the discord for assistance if the issue is still occurring.

closed 2024-04-16 by riptidewave93