X1Plus/X1Plus #174: improve accessibility / iconography for home axis button

jwise, 2024-04-09:
Following up on https://github.com/X1Plus/X1Plus/pull/163#issuecomment-2044273468 .

We could probably make it a little more visually obvious ("home" with exclamation point?  "home" with question mark?) other than just by color that something is wrong, and that would also improve a11y some too.

bani6809, 2024-04-10:
this is my proposal
https://github.com/X1Plus/X1Plus/pull/163#issuecomment-2046479849

jphannifan, 2024-04-16:
Here is the latest implementation. Red with outline = not homed. Solid green = homed. 500ms ColorAnimation triggers when home status changes
https://github.com/X1Plus/X1Plus/assets/149451641/257e9881-790b-4c14-920f-036c3cfb8ba8

bani6809, 2024-04-18:
@jphannifan is this resolved by #163 ?

bani6809, 2024-05-02:
i am calling this resolved by 553b8055c855914fccc1127901c73f867e70ec2f