X1Plus/X1Plus #176: X1Plus settings daemon

jwise, 2024-04-10:
ota daemon is relatively autonomous, gets variable updates from bbl_screen

this is tied with settings synchronization

ota:
  * should I check for OTAs at all, to begin with, for any reason?
  * when should I check for OTAs? (push command, please do it)
  * should I auto download OTAs?
  * I have an OTA, should I download it?

ota:
  Q: should I check for OTAs at all, to begin with, for any reason?
    A: x1plus settings.json has “true” or “false”, set by bbl_screen
      Q: how do I know when this variable changes?
  Q: when should I check for OTAs? (push command, please do it)
    A: x1plus settings.json has “daily”, “hourly”, … set by bbl_screen
  Q: should I auto download OTAs?
    A: x1plus settings.json has “true” or “false”, set by bbl_screen  
  Q: I have an OTA, should I download it if auto download is turned off?
    A: bbl_screen sends a DDS command to “yes go do it k thx”

Settings interface proposal:

  settings live in settings.json (or something), it is always safe to read this
  bbl_screen is the only entity that will write settings.json
    (or the x1plusd, or someone)
  in order to read a setting once:
    just read it from the settings.json
  in order to set a setting:
    you have to ask the owner, with a DDS message: device/x1plus/request -> { “settings”: { “key”: values… } }
  how do you know when a setting has changed?
    any time any setting changes, the owner will broadcast to device/x1plus/report: { “settings_changed”: { “key”: value… } }

  Q: should I check for OTAs at all, to begin with, for any reason?
    A: x1plus settings.json has “true” or “false”, set by bbl_screen
      Q: how do I know when this variable changes?
        A: first, subscribe to device/x1plus/report.  now, read settings.json on startup and see what settings[“ota_enabled”] is.  now, react to device/x1plus/report events with “settings_changed: { “ota_enabled”: true / false }

x1plus-settings.py ota_enabled=true
  # calls up DDS, makes a DDS request
  # only tricky bit: you have to wait until DDS discovery finishes

ota <-> bbl_screen interface:

device/x1plus/report: { “ota”: { “last_check”: timestamp, “version”: { version info }, “downloaded”: boolean, “last_check_failure_reason”: STRING_CONSTANT, “download_percent”: … } }
device/x1plus/request: { “ota”: { “check”: true } } /* check now */
device/x1plus/request: { “ota”: { “download”: “version-string” } }
device/x1plus/request: { “ota”: { “install”: “version-string” } }

bani6809, 2024-04-10:
imo x1plus settings should live in its own x1plus-settings.json and not share the oem config file?

riptidewave93, 2024-04-10:
@bani6809 that's the plan, ill have a branch of the settings DDS service by EOD.

riptidewave93, 2024-04-10:
Work is at https://github.com/X1Plus/X1Plus/tree/riptide/x1plus_settings

jwise, 2024-05-08:
we did it

closed 2024-05-08 by jwise