X1Plus/X1Plus #177: Allow 120C bed temp on 240V power.

jsondag, 2024-04-10:
**Is your feature request related to a problem? Please describe.**
When running on 240V bed temperature is limited to 110C.

**Describe the solution you'd like**
I would like to be able to set bed temp to 120C as I can when I run on 120V.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Software Information**
 - Current X1Plus Version: None Yet

**Additional context**
This is supposedly because of leakage current at 240V, but I'm in the US not using GFCI

RobinWhitfield, 2024-04-13:
@bani6809, are we 100% sure it requires mc firmware modification?

bani6809, 2024-04-18:
```
Apr 17 19:38:47 err device_gate[1000]: [device_topic_outbound_forward:128] pub failed:device/00M00A270400120/report @mqtt
Apr 17 19:38:47 debug bbl_screen: [x1p] Gcode published: {"command":"gcode_line","param":"M140 S120","sequence_id":"420"}
Apr 17 19:38:47 info forward[1036]: [gparser][565] get gcode dds msg: {"command":"gcode_line","param":"M140 S120","sequence_id":"420"}
Apr 17 19:38:47 info forward[1036]: [gparser][176] gcode_line_handle start to handle, src : 0600 content : M1009 L1 O1 M140 S120 M1009 L0 O1
Apr 17 19:38:47 info forward[1036]: [gparser][199] gcode_line_handle send last string is: M1009 L1 O1 M140 S120 M1009 L0 O1  seq is: 7
Apr 17 19:38:47 info forward[1036]: [gparser][569] get gcode dds msg handled
Apr 17 19:38:47 err device_gate[1000]: [device_topic_outbound_forward:128] pub failed:device/00M00A270400120/report @mqtt
Apr 17 19:38:47 info forward[1036]: [MCU][BMC]E-Ist [34]B sus! o=0 s=0
Apr 17 19:38:47 info forward[1036]: [MCU][LINK]GcodeLine (7) l=34 from 0600 ok: M1009 L1 O1 M140 S120 M1009 L0 O1
Apr 17 19:38:47 info forward[1036]: [gparser][33] gcode_line_handle insert succcess, ack seq: 7
Apr 17 19:38:47 info forward[1036]: [MCU][BMC]owner=1 lock! [Free]
Apr 17 19:38:47 err device_gate[1000]: [device_topic_outbound_forward:128] pub failed:device/00M00A270400120/report @mqtt
Apr 17 19:38:47 info forward[1036]: [MCU][BMC]M140:now plate idx:0
Apr 17 19:38:47 info forward[1036]: [MCU][BMC]M140 A:plate unknow,use S
Apr 17 19:38:47 info forward[1036]: [MCU][BMC]M140: Set heatbed to 110.0, input=120.0, exceed limit @volt_220
Apr 17 19:38:47 info forward[1036]: [MCU][BMC]owner=1 unlock!
Apr 17 19:38:47 info forward[1036]: [MCU][HMS]HMS Code = 0x0300-0100-0003-0008
Apr 17 19:38:47 info forward[1036]: [hms][mcu_hms_cb][112] add mcu hms msg attr: 03000100 code: 00030008
Apr 17 19:38:47 debug bbl_screen: BBLPrinter::publishMessage device/inter/report/tracker "{\"content\":{\"chain\":\"pageBtn_Settings - subPageBtn_Maintenance - enterBtn\",\"control\":\"operationChain\"},\"evt\":\"jPrint_UITrack\"}"
Apr 17 19:38:47 debug bbl_screen: BBLPrinter::publishMessage device/inter/report/tracker "{\"content\":{\"control\":\"ZButton\",\"event\":\"clicked\",\"itemName\":\"enterBtn\",\"lang\":\"en\",\"pageName\":\"Console\",\"printStage\":1,\"printState\":0},\"evt\":\"jPrint_UITrack\"}"
Apr 17 19:38:47 info forward[1036]: online node /tmp/device/online is not exsit
```