X1Plus/X1Plus #178: Republish MQTT messages to DDS

jphannifan, 2024-04-11:
To allow us to create custom MQTT commands, we need to implement a listener that runs in the background and when it sees these messages, it publishes them to DDS (or interprets the messages directly). My vote would be to republish messages to DDS, because otherwise we may end up interpreting messages in DDS and in MQTT separately and that sounds like chaos. 

This will allow us to implement new MQTT commands for HA users (and if or when Bambu removes certain privileges from MQTT). This issue here provides more detail on the application https://github.com/X1Plus/X1Plus/issues/69

As of v.1.1 of X1Plus, messaging on the X1 looks like this
[image]

Here's a very rough example to show the two main options it seems we have here:

Lets say we want an MQTT command that can be used to read our settings file and return key values:

1. Listener service that will republish MQTT to DDS
MQTT request "get_setting" --> MQTT listener on printer --> identical DDS request "get_setting" --> DDS response 

2. Listener service that also interprets MQTT messages instead of republishing to DDS
MQTT request "get_setting" --> MQTT listener on printer -->  MQTT response
DDS request "get_setting" --> DDS listener on printer --> DDS response

jwise, 2024-04-14:
I sort of wonder if we can interpose `device_gate` to make it do some of this for us.

jphannifan, 2024-04-14:
> I sort of wonder if we can interpose `device_gate` to make it do some of this for us.

I like this idea a lot, because the alternative is probably using Paho in Python so we can keep DDS callbacks synced with MQTT pubs, and that sounds like a can of worms

jphannifan, 2024-05-12:
We can do this with Dbus not DDS

closed 2024-05-12 by jphannifan