X1Plus/X1Plus #184: X1Plus custom firmware installation download failed

gogsi1, 2024-04-15:
**Describe the bug**
I installed X1 Plus 1.0 on 01.07.02.00, after that I installed update 1.0-7+gca7e1d4.x1 and wanted to install X1Plus1.1rc1, but it gives an error because in this mode the printer does not connect to wifi.

**To Reproduce**
Steps to reproduce the behavior:
1. I downloaded the firmware to the SD card
2. Logged into the x1plus installer
3. I chose the firmware to install X1Plus1.1rc1

**Screenshots/Photos**
[image]

**External Links**
It seems that I found one topic, but there is not much how to fix it there

**Additional Details:**
 - Printer type (X1):

riptidewave93, 2024-04-16:
Hello @gogsi1, can you confirm if your printer is in LAN or Cloud mode? If you can, please boot it into the official firmware (updater disabled) and confirm it is set to Cloud mode. Once done, please reboot and re-try installing the 1.1rc1 update.

If you continue to have issues, it's highly recommended to use the Discord for assistance. https://github.com/X1Plus/X1Plus/wiki#sounds-good-now-what

gogsi1, 2024-04-17:
> Hello @gogsi1, can you confirm if your printer is in LAN or Cloud mode? If you can, please boot it into the official firmware (updater disabled) and confirm it is set to Cloud mode. Once done, [...]

Cloud mode is set

gogsi1, 2024-04-18:
> I had this same issue. After seeing this post I booted to official firmware (updater disabled) to check cloud access which was on. But I noticed in this mode the WiFi was not connected. So I connnected to WiFi while booted to official firmware then restarted. Now it downloads the base firmware just fine.

Thanks, I was able to update.

Turtletrumpet, 2024-04-18:
I run X1plus V1.0, I had this same issue. After seeing this post I booted to official firmware (updater disabled) to check cloud access which was >off<. Wifi was connected. I activate cloud access and power cycle the printer. I start the update process again and no download of base firmware is possible. I see on the wifi AP, that the printer is not connected. Please help. Thank you.

jphannifan, 2024-04-18:
> I run X1plus V1.0, I had this same issue. After seeing this post I booted to official firmware (updater disabled) to check cloud access which was >off<. Wifi was connected. I activate cloud access [...]

I'm looking into this, but just so you know, "Start built-in firmware with updater disabled" boots OEM firmware with LAN mode enabled. So you are activating LAN mode by selecting this option.

bani6809, 2024-04-18:
there should be an x1plus_installer_log.txt in the root of the sdcard, if you could attach it here when you get a failure that would help us debug this problem.

gogsi1, 2024-04-19:
To update from version 01.07.02.00 to 01.07.03.00
Restart the printer in the "X1 Plus Startup Options" section, select "Run the firmware without changes".
After starting, we refuse to update, turn on the local network mode and install via
x1 plus-install-1.1-win32-x64 firmware.

This is the only way I was able to update

jphannifan, 2024-04-19:
> To update from version 01.07.02.00 to 01.07.03.00 Restart the printer in the "X1 Plus Startup Options" section, select "Run the firmware without changes". After starting, we refuse to update, turn [...]

There should be a file on the root of your sd card called `x1plus_installer_log.txt`. Could you please attach that here so we can have a look? 
[image]

Turtletrumpet, 2024-04-19:
> > To update from version 01.07.02.00 to 01.07.03.00 Restart the printer in the "X1 Plus Startup Options" section, select "Run the firmware without changes". After starting, we refuse to update, turn on the local network mode and install via x1 plus-install-1.1-win32-x64 firmware.
> > This is the only way I was able to update
> 
> There should be a file on the root of your sd card called `x1plus_installer_log.txt`. Could you please attach that here so we can have a look? <img alt="image" width="509" src="https://private-user-images.githubusercontent.com/149451641/323893099-07c35508-c1b1-4a8e-80a9-82cf1aacf9f9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM1MjMzNzIsIm5iZiI6MTcxMzUyMzA3MiwicGF0aCI6Ii8xNDk0NTE2NDEvMzIzODkzMDk5LTA3YzM1NTA4LWMxYjEtNGE4ZS04MGE5LTgyY2YxYWFjZjlmOS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDE5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQxOVQxMDM3NTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04ODUzNDkwZDc5YWNhNGJlMDVkMWFhZTEzZWMwOGVmMDAwMjdiYzk3NjQwYjY4OTYwMjRjNjUwOTM0ZTgzNzdmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.nstZyM2dXCYEH0WMbi_9jc3QMjPfuXxUTQP0mtFj3n0">

Hello,
here is my x1plus_installer_log.txt. 
I have tried to install the "base firmware" in "LAN only" mode.
The main problem:
No matter what I try, the X1C does not connect to the WiFi AP during the update process.

[x1plus_installer_log.txt](https://github.com/X1Plus/X1Plus/files/15039349/x1plus_installer_log.txt)

stnby, 2024-04-20:
I had issue when updating to v1.1 from v1.0 that it corrupted the install and failed to boot. Fresh reinstall to empty SD card helped recover.

Interesting part of the log:
```
mount: mounting /dev/loop0 on /tmp/rw_ext4 failed: Invalid argument
[...] Creating writable rootfs on SD card
mke2fs 1.43.9 (8-Feb-2018)
Creating filesystem with 262144 4k blocks and 65536 inodes
Filesystem UUID: 42dbe1aa-9a3f-4155-8165-a0010c650263
Superblock backups stored on blocks: 
	32768, 98304, 163840, 229376

Allocating group tables: 0/8   done                            
Writing inode tables: 0/8   done                            
Creating journal (8192 blocks): done
Writing superblocks and filesystem accounting information: 0/8   done

[+]
[...] Copying custom firmware
packing printer_ui.so
Waiting for remaining data blocks...
Writing inodes and directories...
Writing fragment table...
Writing ID table...
Writing extended attributes...
---------------------------------------------------
Data bytes read: 3.9M
Data bytes written: 2.4M
Data compression ratio: 62%

Data blocks written: 32
Out of which where fragment blocks: 1
Duplicate blocks omitted: 0
Sparse blocks omitted: 0

Fragments actually written: 1
Duplicated fragments omitted: 0
Total number of inodes: 3
Number of unique group/user IDs: 1

[+]
[...] Copying kernel
[+]
[...] Generating boot.conf
[+]
[...] Installing bootloader shim
[...] ... Installing on primary partition. ...
mount: mounting /dev/root on / failed: Device or resource busy
mount: mounting /dev/root on / failed: Device or resource busy
[...] ... Installing on backup partition. ...
[+]
[+] complete
shutting down Python DDS thread
```

jwise, 2024-04-20:
@stnby Can you say what the failure was when it failed to boot later?

gogsi1, 2024-04-21:
> There should be a file on the root of your sd card called `x1plus_installer_log.txt`. Could you please attach that here so we can have a look? <img alt="image" width="509" [...]

[x1plus_installer_log.txt](https://github.com/X1Plus/X1Plus/files/15050778/x1plus_installer_log.txt)

Turtletrumpet, 2024-04-22:
Hello,
the only way i found to update from V1.0 to V1.1 is to boot in the unmodified original firmware.
Then I use the X1Plus Installer on my PC to update. It works fine and i'm happy.

jinnn1989, 2024-04-27:
Facing similar issues while installing 1.1, Base firmware download failed

RABBIT-92, 2024-04-30:
> > Also having base firmware download failures. I tried every fix in this thread but it still fails. I think I'm just gonna stick with 1.0 and move on before I get frustrated.
> 
> We're working on a fix. There appears to be more than just one issue here, but yours looks like our Wi-Fi driver issue
> 
> You can download the update from Bambu and put the file on your sd card and this should allow you to update. Don't change the filename and make sure you place it on the root of your sd card: /mnt/sdcard/
> 
> [update-v00.00.32.17-20240409222723_product.img.zip.sig](https://public-cdn.bambulab.com/upgrade/device/BL-P001/01.07.03.05/product/79a426f63e/update-v00.00.32.17-20240409222723_product.img.zip.sig)

I downloaded and placed it in the root directory and it shows this failure message

[image]

jphannifan, 2024-04-30:
> > > Also having base firmware download failures. I tried every fix in this thread but it still fails. I think I'm just gonna stick with 1.0 and move on before I get frustrated.
> 
> > 
> 
> > We're working on a fix. There appears to be more than just one issue here, but yours looks like our Wi-Fi driver issue
> 
> > 
> 
> > You can download the update from Bambu and put the file on your sd card and this should allow you to update. Don't change the filename and make sure you place it on the root of your sd card: /mnt/sdcard/
> 
> > 
> 
> > [update-v00.00.32.17-20240409222723_product.img.zip.sig](https://public-cdn.bambulab.com/upgrade/device/BL-P001/01.07.03.05/product/79a426f63e/update-v00.00.32.17-20240409222723_product.img.zip.sig)
> 
> 
> 
> I downloaded and placed it in the root directory and it shows this failure message. I also tried putting it in the directory the error message specified but it also didn't work. It just asks for a different base firmware version. 
> 
> 
> 
> [image]
> 
> 

I told you the root of the SD card but I should have double checked first. The correct path for firmware files to successfully skip the download in this version of x1plus is 
`/mnt/sdcard/x1plus/firmware/`

RABBIT-92, 2024-04-30:
> > > > Also having base firmware download failures. I tried every fix in this thread but it still fails. I think I'm just gonna stick with 1.0 and move on before I get frustrated.
> > 
> > 
> > > 
> > 
> > 
> > > We're working on a fix. There appears to be more than just one issue here, but yours looks like our Wi-Fi driver issue
> > 
> > 
> > > 
> > 
> > 
> > > You can download the update from Bambu and put the file on your sd card and this should allow you to update. Don't change the filename and make sure you place it on the root of your sd card: /mnt/sdcard/
> > 
> > 
> > > 
> > 
> > 
> > > [update-v00.00.32.17-20240409222723_product.img.zip.sig](https://public-cdn.bambulab.com/upgrade/device/BL-P001/01.07.03.05/product/79a426f63e/update-v00.00.32.17-20240409222723_product.img.zip.sig)
> > 
> > 
> > I downloaded and placed it in the root directory and it shows this failure message. I also tried putting it in the directory the error message specified but it also didn't work. It just asks for a different base firmware version.
> > ![20240430_115642](https://private-user-images.githubusercontent.com/8989932/326889335-4fb9455c-a122-4315-80da-d22fd718d50a.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ1MTE0MDQsIm5iZiI6MTcxNDUxMTEwNCwicGF0aCI6Ii84OTg5OTMyLzMyNjg4OTMzNS00ZmI5NDU1Yy1hMTIyLTQzMTUtODBkYS1kMjJmZDcxOGQ1MGEuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQzMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MzBUMjEwNTA0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NjAzOThiNGMyOTM1NmRkODJmY2M2YjM3MzI0NzhiM2Y3ZGU4YTBmZGI4ZGRkNjk5ZmVkY2NhYWY1OGYwMGMxMiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.pa10KTLpJWtaFm-wLBwTk8f1zCRsBeew66OkpC2-Dhg)
> 
> I told you the root of the SD card but I should have double checked first. The correct path for firmware files to successfully skip the download in this version of x1plus is `/mnt/sdcard/x1plus/firmware/`

It throws the same error when placing in that directory as well. It's asking me for an older firmware version than what you linked earlier.

jphannifan, 2024-04-30:
> > > > > Also having base firmware download failures. I tried every fix in this thread but it still fails. I think I'm just gonna stick with 1.0 and move on before I get frustrated.
> 
> > > 
> 
> > > 
> 
> > > > 
> 
> > > 
> 
> > > 
> 
> > > > We're working on a fix. There appears to be more than just one issue here, but yours looks like our Wi-Fi driver issue
> 
> > > 
> 
> > > 
> 
> > > > 
> 
> > > 
> 
> > > 
> 
> > > > You can download the update from Bambu and put the file on your sd card and this should allow you to update. Don't change the filename and make sure you place it on the root of your sd card: /mnt/sdcard/
> 
> > > 
> 
> > > 
> 
> > > > 
> 
> > > 
> 
> > > 
> 
> > > > [update-v00.00.32.17-20240409222723_product.img.zip.sig](https://public-cdn.bambulab.com/upgrade/device/BL-P001/01.07.03.05/product/79a426f63e/update-v00.00.32.17-20240409222723_product.img.zip.sig)
> 
> > > 
> 
> > > 
> 
> > > I downloaded and placed it in the root directory and it shows this failure message. I also tried putting it in the directory the error message specified but it also didn't work. It just asks for a different base firmware version.
> 
> > > ![20240430_115642](https://private-user-images.githubusercontent.com/8989932/326889335-4fb9455c-a122-4315-80da-d22fd718d50a.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ1MTE0MDQsIm5iZiI6MTcxNDUxMTEwNCwicGF0aCI6Ii84OTg5OTMyLzMyNjg4OTMzNS00ZmI5NDU1Yy1hMTIyLTQzMTUtODBkYS1kMjJmZDcxOGQ1MGEuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQzMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MzBUMjEwNTA0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NjAzOThiNGMyOTM1NmRkODJmY2M2YjM3MzI0NzhiM2Y3ZGU4YTBmZGI4ZGRkNjk5ZmVkY2NhYWY1OGYwMGMxMiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.pa10KTLpJWtaFm-wLBwTk8f1zCRsBeew66OkpC2-Dhg)
> 
> > 
> 
> > I told you the root of the SD card but I should have double checked first. The correct path for firmware files to successfully skip the download in this version of x1plus is `/mnt/sdcard/x1plus/firmware/`
> 
> 
> 
> It throws the same error when placing in that directory as well. It says it's looking for a different firmware version than what you linked earlier: v00.00.28.59-20240330174856_products.img.zip.sig

Here's the link for that specific version. I should not answer GitHub issues on mobile 🙂

[update-v00.00.28.59-20240330174859_product.img.zip.sig](https://public-cdn.bambulab.com/upgrade/device/BL-P001/01.07.03.00/product/1916138d2c/update-v00.00.28.59-20240330174859_product.img.zip.sig)

RABBIT-92, 2024-04-30:
> Here's the link for that specific version. I should not answer GitHub issues on mobile 🙂 [...]

Hell yeah, it finally let me update. Thank you!

yunginnanet, 2024-05-05:
### Initial failure

`install.py` kept getting an HTTP 403 in the log when using requests to fetch the firmware on the printer (using normal installer from releases page):

```
[...] ... Connecting... ...
/userdata/x1plus/install.py:280: DeprecationWarning: cafile, capath and cadefault are deprecated, use a custom context instead.
  with urllib.request.urlopen(basefw_update_url, capath="/etc/ssl/certs") as resp, \
... failed (Download failed.)
shutting down Python DDS thread
Traceback (most recent call last):
  File "/userdata/x1plus/install.py", line 295, in <module>
    report_failure(f"Download failed.", e)
  File "/userdata/x1plus/install.py", line 75, in report_failure
    raise e
  File "/userdata/x1plus/install.py", line 280, in <module>
    with urllib.request.urlopen(basefw_update_url, capath="/etc/ssl/certs") as resp, \
  File "/userdata/x1plus/lib/python3.10/urllib/request.py", line 216, in urlopen
    return opener.open(url, data, timeout)
  File "/userdata/x1plus/lib/python3.10/urllib/request.py", line 525, in open
    response = meth(req, response)
  File "/userdata/x1plus/lib/python3.10/urllib/request.py", line 634, in http_response
    response = self.parent.error(
  File "/userdata/x1plus/lib/python3.10/urllib/request.py", line 563, in error
    return self._call_chain(*args)
  File "/userdata/x1plus/lib/python3.10/urllib/request.py", line 496, in _call_chain
    result = func(*args)
  File "/userdata/x1plus/lib/python3.10/urllib/request.py", line 643, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 403: Forbidden
```

### Workaround

 i ssh'd into the printer, explored around `/userdata/x1plus`, ran `pkill -i -e python`, and used `vi` to modify `install.py` to just use wget:
 
 ```diff
282a278
>             print(basefw_update_path)
284,297c280,283
<             report_interim_progress("Connecting...")
<             with urllib.request.urlopen(basefw_update_url, capath="/etc/ssl/certs") as resp, \
<                  open(basefw_update_path, "wb") as f:
<                 totlen = resp.getheader('content-length')
<                 curlen = 0
<                 while True:
<                     buf = resp.read(131072)
<                     if not buf:
<                         break
<                     f.write(buf)
<                     curlen += len(buf)
<                     curlen_mb = round(int(curlen) / 1024 / 1024, 2)
<                     totlen_mb = round(int(totlen) / 1024 / 1024, 2)
<                     report_interim_progress(f"Downloading... ({curlen_mb:.2f} / {totlen_mb:.2f} MB)") 
---
>             report_interim_progress("Downloading with wget...")
>             wgcmd = f"wget --output-document {boot_path}/firmware/{basefw_update} {basefw_update_url}"
>             ret = os.popen(wgcmd).read()
>             print(ret)
```

### Success

[image]

jphannifan, 2024-05-05:
> ### Initial failure
> `install.py` kept getting an HTTP 403 in the log when using requests to fetch the firmware on the printer (using normal installer from releases page):
> 

> ```
The 403 error is due to urllib/requests not having a user agent header. The "Download failed" is a separate wifi issue that I'm still working on. I have a branch with both of these issues nearly fixed but I suspect there are race conditions that will drag this issue out further than necessary. 

[1.1_rc1-30+gbb0398e.x1p](https://github.com/X1Plus/X1Plus/releases/download/v1.0-7-brcmfmac/1.1_rc1-30+gbb0398e.x1p) is my partial fix. I'll be updating this page with the full fix when I have sorted this out. 

https://github.com/X1Plus/X1Plus/releases/tag/v1.0-7-brcmfmac

Thanks for sharing this info.

jphannifan, 2024-05-05:
We have a fix ready for this. If anyone encounters difficulties with this update, please let me know. After installing this, your boot screen should display a wifi icon in the top right corner of the screen. You will not see it until after a successful install, though.

https://github.com/X1Plus/X1Plus/releases/download/v1.0-7-brcmfmac/1.1_rc1-33+g2d382aa.x1p

jphannifan, 2024-06-09:
We have fixed this issue (PR 239) and made some extensive changes to the updater to enable OTA updates, so it's unlikely to run into this bug unless running an old version of X1Plus. OTA updates will be available in v2.0

closed 2024-06-09 by jphannifan