X1Plus/X1Plus #187: UI button to enable perf_logging for 8 hours

bani6809, 2024-04-16:
UI button somewhere that enables perf_logging for 8 hours

jwise, 2024-04-16:
Another thought I had in the general genre of “debugging the tool head stall problem”: in addition to the speed adjust screen, I wonder if we  could have a speed adjust mode that sets the feed rate to 0, so we could “virtually print” on bad configurations, but at least not waste plastic.

jphannifan, 2024-04-16:
> Another thought I had in the general genre of “debugging the tool head stall problem”: in addition to the speed adjust screen, I wonder if we could have a speed adjust mode that sets the feed rate to 0, so we could “virtually print” on bad configurations, but at least not waste plastic.

I'll test this