X1Plus/X1Plus #188: X1Plus 1.1 rc1 installer JavaScript error

jurjevic, 2024-04-16:
**Describe the bug**
Using the x64 installer on M1 Mac (aarch64) produces a JavaScript error:
`Uncaught Exception:
Error: bind EADDRINUSE 0.0.0.0:2021
at node:dgram:362:20
at process.processTicksAndRejections (node:internal/process/task_queues:83:21)`
[X1Plus 1.1 rc1](https://github.com/X1Plus/X1Plus/releases/tag/x1plus%2F1.1_rc1)

**To Reproduce**
Steps to reproduce the behavior:
1. Download installer
2. Execute installer

**Expected behavior**
no error

**Screenshots/Photos**
[image]

**External Links**

**Additional Details:**

**Is the bug present on OEM firmware?**

**Custom modifications**

**Additional context**
- Shouldn't be a aarch64 installer provided for new Macs?

riptidewave93, 2024-04-16:
CC @jwise

jwise, 2024-04-17:
This is probably the issue where a slicer is running. We should have a better error for this though.

jwise, 2024-04-17:
Closing as dup of #54 .