X1Plus/X1Plus #191: Region lock printers

g992, 2024-04-18:
hello! does the region block work on x1 plus? I want to use chinese x1 outside of china without vpn

jphannifan, 2024-04-18:
> hello! does the region block work on x1 plus? I want to use chinese x1 outside of china without vpn

So you need the printer to display text in Chinese? I think you can do this already, but let me check on that and get back to you.

bani6809, 2024-04-18:
LAN mode works so you can use any X1 in any country.

g992, 2024-04-19:
> > hello! does the region block work on x1 plus? I want to use chinese x1 outside of china without vpn
> 
> 
> 
> So you need the printer to display text in Chinese? I think you can do this already, but let me check on that and get back to you.

I want to use Chinese printer with  eu bambu handy app. but as I understand it, this is impossible, and the only option is to use it in “lan only” mode

bani6809, 2024-04-19:
why do you have a chinese x1? if you want to use bambu cloud in eu, you need to buy an eu x1.

region code is registered with the printer Serial Number and x1plus cannot change it.

psiberfunk, 2024-04-19:
To be clear : even if it could , the X1Plus team is not interested in subverting region restrictions and encouraging potential resale avenues that would undermine normal legitimate distribution ones.  Not only would this be ethically murky at best, but it would also open up legal liability issues that we don’t want to get into. 

If you somehow have the wrong region printer through proper channels , I suggest you contact your retailer.  If you’ve purchased a printer for cheap via grey market sellers with the expectation that Bambu’s services should work.. then sorry , this is not the place for that.

closed 2024-04-19 by psiberfunk