X1Plus/X1Plus #194: Followup Screen Lock bug - Unable to enter password

Johnnybyzhang, 2024-04-21:
**Describe the bug**
This is a follow-up for #126 As the approach that fixed the keypad overlay issue has worsened it. When I wake the printer from sleep by pressing the power button or tapping on the screen, the password keypad then shows up, except that I am now unable to enter a passcode. I will attach a video showing this very issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Have a password set and update to X1Plus Version 1.1
2. Press the power button to wait to put the printer to sleep
3. Wake the printer from sleep by pressing the power button/tapping the screen
4. The keypad will show up and one is unable to enter the password

**Expected behavior**
The lock screen should show up, having the password screen should also be an option (if applicable), but having the keypad while not being able to use it is clearly incorrect.

**Screenshots/Photos**

https://github.com/X1Plus/X1Plus/assets/42496003/d1bd9ed0-5d64-44ec-90c8-19c814e392d7

**External Links**
This is a follow-up to #126 which is now closed but I believe it is still related.

**Additional Details:**
 - Printer type (X1/X1C/X1E): X1C
 - X1Plus version: 1.1
 - AP firmware version: 00.00.28.59
 - AP hardware revision: AP05
 - MC firmware version: 00.00.22.41
 - MC hardware revision: MC07
 - Toolhead firmware version: 00.00.07.12
 - Toolhead hardware revision: TH09
 - Computer vision model: 

**Is the bug present on OEM firmware?**
Have you tested whether this bug is also present in OEM firmware?
If it is, which version? *ex. 01.07.02.00*
Are any steps to replicate it different?

Not possible in OEM

**Custom modifications**

None

**Additional context**
Update: I have restarted the screen service in an attempt to get snapshots, but it seems to have fixed the issue. I can still help by providing logs.

jwise, 2024-04-21:
Whoah, interesting. Thanks for the detailed report. Is this the first time you’ve seen this, or does it happen regularly? I will stare harder at the code for this to see if I can understand how this could happen.

Johnnybyzhang, 2024-04-21:
Update: The reproduction method is the same as in #126, spamming the swipe to unlock is the reason to this very bug, just that it have a smaller impact

Johnnybyzhang, 2024-04-21:
https://github.com/X1Plus/X1Plus/assets/42496003/579a1868-b99b-4d35-a477-c297f957bc7a

PR #202 (2024-04-24): ScreenLock: take an extremely belt-and-suspenders approach to making sure that we do not pop a box multiple times (fixes #194)
There were a handful of ways that the screen lock could go wrong.  We shore up many of them:

* The StackView apparently did not work at all when we tried to `pop` from it.  We give it a dummy item so that it is permitted to `pop`.
* If you somehow manage to tap behind the number pad without dismissing it, it would be good to make sure that the thumb is not draggable.
* If you somehow manage to tap the number pad and drag the thumb, it would be good to avoid popping th [... cut]
commit 20ca3a7e (2024-04-24): ScreenLock: take an extremely belt-and-suspenders approach to making sure that we do not pop a box multiple times (fixes #194) (#202)

There were a handful of ways that the screen lock could go wrong. We shore up many of them:

* The StackView apparently did not work at all when we tried to pop from it. We give it a dummy item so that it is permitted to pop.
* If you somehow manage to tap behind the number pad without dismissing it, it would be good to make sure that the thumb is not draggable.
* If you somehow manage to tap the number pad and drag the thumb, it would be good to avoid popping  [... cut]
closed 2024-04-24 by jwise