X1Plus/X1Plus #197: 1.1 wifi drivers do not support DE region channels

bani6809, 2024-04-22:
it appears the 1.1 wifi drivers do not support wifi channels 12,13 (DE)

likely missing from the murata regulatory clm.

this is probably the cause of issues #184 and other reports on the discord server.

jwise, 2024-04-22:
Remind me how we decided it is the firmware and not brcmfmac?

bani6809, 2024-04-22:
> Remind me how we decided it is the firmware and not brcmfmac?

as much as I would have liked to blame brcmfmac, i was systematic and ruled it out by having Digilol on discord repro their bug with these builds: https://github.com/X1Plus/X1Plus/releases/tag/v1.0-7-brcmfmac

one is murata, one is murata+brcmfmac

bug repro'd in both. and in both cases user resolved their problem by changing from channel 13 to another channel.

bani6809, 2024-05-02:
should be fixed by 0f906c3e4b7e9af948d2bcc5132666a3a23426fe

closed 2024-05-02 by bani6809