X1Plus/X1Plus #201: Update Firmware Section in the User Manual

reipet, 2024-04-23:
**Is your feature request related to a problem? Please describe.**
I recently found in the manual the firmware page is described to show the current firmware running on the machines modules.
Since update to x1Plus 1.1. there is a message the MC firmware version is not compatible.
When I press update from the firmware page it wants to connect to Bambu Web-Servers but nothing happens then anymore.
So we dont know what failed here, it is just stuck in "Connecting"

**Describe the solution you'd like**
Add a description if it is safe to press update the firmware other then the X1Plus from the firmware page or other things need to be considered beforehand.
Is it possible to normally update the other modules firmware with the X1 Plus firmware running on the machine?

**Software Information**
 - Current X1Plus Version: 1.1

bani6809, 2024-04-24:
@reipet unless you have LAN+SHIELD enabled, the cause of this is almost certainly due to bug #197

reipet, 2024-05-23:
This is correct, I was in the wrong mode. Maybe a hint in the manual would help users like me :-)