X1Plus/X1Plus #208: 1.1 update missing run macro on button dropdown

BusteanHAN, 2024-04-24:
**Describe the bug**
In the dropdown to change the action of a button, the "Run Macro" option, that is still detailed by the question mark help, is not present.
[image]

I have run the x1p firmware version 1.0 since it came out and had bound the On/Off button to a python macro for nozzle changes, that became unbound after updating to 1.1

I noticed when ssh-ing that inside the folder of the printer there is a buttons.json, I guess that can be edited to add back the macro, but the editing of this file does not seem to be documented yet.

BusteanHAN, 2024-04-25:
Thank you for the explanation, this was the kind of workaround I expected tbh, I even cloned and tried to build, but got tired setting up docker on windows partway and never really got to it again, I hope one day I can contribute to the project in some way, but it all looks beyond me right now.

BusteanHAN, 2024-04-27:
Appreciate the suggestion for tools, I am familiar tho with json :))

It took me a while to find where S82gpiokeys was, and found it in /etc/init.d in order to not have to restart the printer each time since it takes a while too boot

Despite all that, having replaced the py script with the one you mentioned, and having looked in the filesystem a bit more, I cannot find the reason why this is not working

```json
{
    "power": {
        "shortPress": {
            "action": "ACTION_MACRO",
            "parameters": "/mnt/sdcard/x1plus/macros/nozzleChange.py"
        },
        "longPress": {
            "action": "ACTION_TOGGLE_SCREENSAVER",
            "parameters": {}
        }
    },
    "estop": {
        "shortPress": {
            "action": "ACTION_ABORT",
            "parameters": {}
        },
        "longPress": {
            "action": "ACTION_ABORT"
        }
    }
}
```
Above is my buttons.json
I replaced the .py file in a number of ways, first by using wget <raw github link> -o gpiokeys.py, then copying it to /opt, then by just editing the file and copy pasting the code, just to make sure I am not insane and I missed something.

Checking S82gpiokeys in init.d, the paths in the script correspond with where my files are arranged

And despite all that, rebooting the printer a number of times, and doing ./etc/init.d/S82gpiokeys restart, having the screen show the option "Choose" for the action
[image]

The power button still performs the default action, of sleeping the screen

P.S. if you need me to send over more info, please tell me; I did not want to bloat this comment more than I already have

jphannifan, 2024-04-27:
Ohh I'm sorry. this is my fault. The script I shared calls this on line 67:
`subprocess.run(["python3", script_path], check=True)`

However the version of X1Plus you're running doesn't have python3 added to the PATH variable, so the full filepath should have been provided. 

Creating this symlink should fix the script for you so you don't have to make any modifications:
`ln -s /opt/python/bin/python3.10 /usr/bin/python3`

If that doesn't fix it, then you can get console output from gpiokeys.py that will tell you the issue by
`python3 /opt/gpiokeys.py`

BusteanHAN, 2024-04-28:
Hey, I just got around to testing it, adding the symlink seems to have not done anything, and after it I tried to runn gpiokeys with the following result:
[image]

```py
from logger.custom_logger import CustomLogger

...

gpio_log = CustomLogger("gpiokeys", "/tmp/gpiokeys.log", 500000, 1, True)
```
That is the line in question, but when checking the logger __init__
```py
class CustomLogger:
    def __init__(self, name, filename, maxBytes=1048576, backupCount=5):
```

The `True` seems to be passed as an extra param, and removing it from the script fixes the issue, the button works

BusteanHAN, 2024-04-28:
As I say that, while the script worked and the printer performed the movements, the screen is doing this now:

https://github.com/X1Plus/X1Plus/assets/60746455/f026707b-52e0-41cf-a879-84022d224f70

How did changing the button script in this way make the screen behave like that? If it's even related

jphannifan, 2024-04-28:
> Hey, I just got around to testing it, adding the symlink seems to have not done anything, and after it I tried to runn gpiokeys with the following result: ![image](https://private-user-images.githubusercontent.com/60746455/326237168-6f879cdc-e0ab-48b1-b00c-040db54b26eb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQyOTE3MTYsIm5iZiI6MTcxNDI5MTQxNiwicGF0aCI6Ii82MDc0NjQ1NS8zMjYyMzcxNjgtNmY4NzljZGMtZTBhYi00OGIxLWIwMGMtMDQwZGI1NGIyNmViLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDI4VDA4MDMzNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTYzNWQ4NDM0MzgwZmRmZDAxNWY3ZmNiYTEwNWU1OWM4Y2FkY2FkZTllZjE4OGFjMzMwM2JlYjMyMjI0MWJkYWImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.gIf4iuE5JDj0EpxjKjZxFcHZ6phFbcml2QTxNrtA-jU)
> 
> ```python
> from logger.custom_logger import CustomLogger
> 
> ...
> 
> gpio_log = CustomLogger("gpiokeys", "/tmp/gpiokeys.log", 500000, 1, True)
> ```
> 
> That is the line in question, but when checking the logger **init**
> 
> ```python
> class CustomLogger:
>     def __init__(self, name, filename, maxBytes=1048576, backupCount=5):
> ```
> 
> The `True` seems to be passed as an extra param, and removing it from the script fixes the issue, the button works

You can comment the logger line out. it's for debugging anyway. I did not realize this script contained so many regressions from v 1.1, so my bad on that! The logger class has changed slightly since the official release of v1.1 and it now expects an additional parameter. Best to just comment it out unless you need the log.

jphannifan, 2024-04-28:
I've seen similar UI behavior in the past, but never one directly related to gpiokeys.py.

You can actually detach the buttons from the UI completely if you want. Give gpiokeys.py a custom path to read settings from and remove the lines where it publishes DDS messages. If you keep experiencing these issues let me know though. 

And just to clarify especially to anyone reading, these are all temporary solutions. You're welcome to modify these scripts as you like. you already have the Python skills to

BusteanHAN, 2024-04-28:
I might try that, having looked into the scripts it looks pretty simple, the only thing is that there is a lot of stuff that I just don't know about the system. One of these days when I have time for it, I might dive deeper into this stuff and hopefully get some useful PRs done;

P.S. For context, I work as junior embedded systems developer, working on air suspension ECUs, this kind of project is right up my alleyway