X1Plus/X1Plus #209: <your bug name here>

Majestic7979, 2024-04-24:
**Describe the bug**
I can keep telling the bed to go down, even when it's at the bottom, which causes grindind noises to occur because the bed can't get any lower. This should not happen.

**To Reproduce**
Steps to reproduce the behavior:
Go to temperature/axis screen. Keep pressing the lower bed button... on normal firmware it will tell you to home. On X1Plus it will simply keep lowering the bed. When the bed gets to the bottom on the normal firmware the bed cannot be lowered any further, the button will not make the motor turn. On X1Plus the motor will turn and you will hear a grinding noise, because the bed cannot move any further, this should not happen. Once it reaches the bottom, it should immediately stop. I believe the end stop is based on the motor, so why is this not being read out and the end stop respected by X1Plus?

**Expected behavior**
When the motor stalls, stop trying to move the bed. It does not use switch end stops so I believe it uses motor torque to decide when it has reached the end. It should respect the fact it can't go any further and stop trying to move, even if the button is being pressed down several times.

**Screenshots/Photos**
If applicable, add screenshots or photos to help explain your problem.

**External Links**
Has this bug been discussed elsewhere that you can link to? Have you checked this repository's discussions section?

**Additional Details:**
 - Printer type (X1/X1C/X1E): X1C
 - X1Plus version: 1.1
 - AP firmware version: 00002859
 - AP hardware revision: ap05
 - MC firmware version: 00002241
 - MC hardware revision: mc07
 - Toolhead firmware version: 00000712
 - Toolhead hardware revision: th07
 - Computer vision model: I do not know where to get this.

**Is the bug present on OEM firmware?**
It does not happen on normal fw

Have you tested whether this bug is also present in OEM firmware?
If it is, which version? *ex. 01.07.02.00*
Are any steps to replicate it different?

**Custom modifications**
Add details of any software or electronic modifications you've made to your printer here.
factory stock, no mods

**Additional context**
Add any other context about the problem here.

psiberfunk, 2024-04-24:
Did you home your bed first?  If not, then you're in homing-unlocked mode.  This is how OEM knows where to stop.  We have homing unlocked by default as a convenience feature.

Majestic7979, 2024-04-25:
> Did you home your bed first? If not, then you're in homing-unlocked mode (X1+ default after boot). This is how OEM knows where to stop. We have homing unlocked by default as a convenience feature. You will note that OEM won't let you do this because it forces you to home first. This can be inconvenient for various different reasons where you may not want to home first.
> 
> Please update this ticket with info on if this occurs when homing first or not.

I did not home first. I prefer the OEM behavior. But I default to loading your firmware. I don't want to juggle between firmwares. I want your firmware to auto load, that's why I agreed to forfeit my warranty. I appreciate the thought behind the convenience but I want to decide how to deal with the Z endstop situation, is there a config file where I can turn this "convenience" off to avoid damaging my z motor/bed? Alternatively is it not possible to simply read the motor parameters and when it stalls you know it has hit an obstacle (such as the bottom) and will stop allowing the button to be pressed? Because think with me here for a second... If the motor is finding an obstacle why would you want to keep it running? What if a child keeps pressing the button over and over just to hear the "funny noise"? I actually like the convenience of not having to home. But I want the firmware to recognize when the motor can't turn freely (has hit the bottom) and will disable further presses until it moves up again. It's acceptable to get a little bit of that grinding noise because how else will the firmware know without homing? But once the motor has shown the firmware it can't move freely, the firmware should recognize that as the end of the travel and disable further moves in that direction (down in this case).

What do you think?

psiberfunk, 2024-04-25:
> not possible to simply read the motor parameters and when it stalls you know it has hit an obstacle (such as the bottom) and will stop allowing the button to be pressed? Because think with me here [...]

"Reading the motor parameters" I believe would involve getting into reverse engineering the MC firmware, which we have not done/do not have control over (yet).  This is a much more complex/opaque system component for which the source code and internals are less clear to us as outsiders, and accordingly, doing things like sense detection is not something that I believe is immediately available to us.

As of the latest firmware, I believe when we are unhomed the home icon warns you visually that this is the case, per #163 

It could in theory be an option to have the "default state on boot" be locked as a configuration option.  However that's not presently something that's implemented.

P.S.  Hopefully you don't let children play with the printer unsupervised ;)

jphannifan, 2024-04-25:
In addition to displaying the home status, #163 added a warning message that's displayed if you try to use the axis controls without homing first.

jphannifan, 2024-04-25:
This issue can occur on OEM too. There are no sensors preventing the motor from stalling

Majestic7979, 2024-04-25:
Many thanks for the kind and detailed reply, it's so different than what I'm used to on github :)

I did not get any message or different icon, I can restart the printer and just press the bed down button and it will go down forever if I keep pushing it repeatedly... I downloaded the installer and installed it yesterday evening, has anything changed since then? I don't mean to be clueless, it's the first time I'm using this firmware so I want to be extra careful to avoid damaging my machine.

You see, when I was a kid my parents tried their best to stop me from doing stupid stuff, but I still did it anyway lol. So while I take precautions, I also keep in mind I was a child one day and when I was curious I was often successful in bypassing restrictions. I just want to make sure I can do all I can to avoid damaging the printer in case I forget to lock the door. It's not your responsibility if something goes wrong but it would be lovely to have a choice.

jwise, 2024-04-25:
I agree with @psiberfunk that if you are childproofing your printer, you really should use our lock screen mode with a password instead :-) You can set the timeout as short as a minute.  But in general, the printer is a 20,000 m/s^2 robot hot glue gun that hits temperatures of 250+C and has a large flat surface in it that hits 110C, and is inherently not a childsafe device!

That said, I also agree with you now that the Z axis homing warning is probably worth putting back.  I removed it before 1.0 because it bothered me, especially when I was doing 'weird' things with my printer (like my vinyl cutter mod).  Internally, we were wondering whether that was really the right thing to do, and basically, I didn't want to give that up, but we also didn't want to write new UI before 1.0 because we had a release to make!

Indeed, @jphannifan fixed this in #163 with the right answer, which is a warning that you can ignore until reboot if you want to do stupid things (like me). I believe this will solve your problem in a future release.

Majestic7979, 2024-04-26:
> I agree with @psiberfunk that if you are childproofing your printer, you really should use our lock screen mode with a password instead :-) You can set the timeout as short as a minute. But in [...]

Thanks, I will wait for the release. ETA please if possible? And if the printer is basically running linux, why is it not possible to code an "update" button on the printer UI and let it connect to github, pull changes and do all the stuff from the printer screen? The update process right now seems cumbersome considering the printer is running linux with root so it is literally an unlocked computer, right? :) Having a little badge icon on top of the settings icon to indicate a new update and then an update button that does it all with 1 click (provided SD has enough space) should be fantastic addition to keep things easy.

jwise, 2024-04-26:
Many things are possible!  Such an 'update' button is possible as well.  We would like to do it but we are all volunteers and have very limited free time.  Join us in #cfw-development on Discord if you'd like to contribute :)

Majestic7979, 2024-04-26:
> Many things are possible! Such an 'update' button is possible as well. We would like to do it but we are all volunteers and have very limited free time. Join us in #cfw-development on Discord if you'd like to contribute :)

I don't know how to code (dyscalculia/maths) but sending love and thanks for what you guys have done so far!

closed 2024-08-30 by jphannifan