X1Plus/X1Plus #226: Unable to install X1Plus

micahfocht, 2024-04-30:
**Describe the bug**
Installer always fails at the "Running on-printer installation" step, particularly after asking to download 

**To Reproduce**
Steps to reproduce the behavior:
1) Walk through the installer application
2) Fail to download firmware on printer.

**Expected behavior**
I expect the installer to download the relevant firmware files.

**Screenshots/Photos**
[image]

`Traceback (most recent call last):
  File "/userdata/x1plus/install.py", line 295, in <module>
    report_failure(f"Download failed.", e)
  File "/userdata/x1plus/install.py", line 75, in report_failure
    raise e
  File "/userdata/x1plus/install.py", line 280, in <module>
    with urllib.request.urlopen(basefw_update_url, capath="/etc/ssl/certs") as resp, \
  File "/userdata/x1plus/lib/python3.10/urllib/request.py", line 216, in urlopen
    return opener.open(url, data, timeout)
  File "/userdata/x1plus/lib/python3.10/urllib/request.py", line 525, in open
    response = meth(req, response)
  File "/userdata/x1plus/lib/python3.10/urllib/request.py", line 634, in http_response
    response = self.parent.error(
  File "/userdata/x1plus/lib/python3.10/urllib/request.py", line 563, in error
    return self._call_chain(*args)
  File "/userdata/x1plus/lib/python3.10/urllib/request.py", line 496, in _call_chain
    result = func(*args)
  File "/userdata/x1plus/lib/python3.10/urllib/request.py", line 643, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 403: Forbidden`

**External Links**
I have not seen discussion of this anywhere.  I saw similar discussion of slicers causing issues, but I ensured none were running and have what appears to be a different error.

**Additional Details:**
 - Printer type (X1/X1C/X1E): X1C
 - X1Plus version: I have attemped 1.0, 1.1 RC, and 1.1
 - AP firmware version: I'm not sure how to see all this on stock firmware.  It's entirely possible I'm just unaware.
 - AP hardware revision:
 - MC firmware version:
 - MC hardware revision:
 - Toolhead firmware version:
 - Toolhead hardware revision:
 - Computer vision model:

**Is the bug present on OEM firmware?**
Bug is present on OEM firmware, as I have not been able to 

**Custom modifications**
None that should affect this.  Added insulation and lights powered externally.

**Additional context**
I have the x1plus_installer_log.txt if that's helpful.  Holding off on posting it publicly in case I need to redact portions of it.

jwise, 2024-04-30:
Can you DM me with the x1plus_installer_log.txt on Discord?  Or e-mail it to me.  Or something :-)

I sort of wonder if there is something weird about your network configuration.  Are you behind a captive portal or something?

micahfocht, 2024-04-30:
> Can you DM me with the x1plus_installer_log.txt on Discord?  Or e-mail it to me.  Or something :-)
> 
> 
> 
> I sort of wonder if there is something weird about your network configuration.  Are you behind a captive portal or something?

I'm not behind a captive portal. I do run pihole for dns but I disabled blocking and attempted to download the file and nothing changed. I also attempted to manually download the file from my computer and the file did download, but putting it in the firmware folder on the sdcard still didn't work. I'll dm you the log shortly.

Alaric323, 2024-04-30:
I'm having a similar issue, no pihole or "unusual" firewall setups.  Either through the PC guided installer or direct from the SD card, the installer for 1.1 is not downloading the base firmware package.  Oddly enough it seems to be trying to download update-v00.00.28.59-20240330174859_product.img.zip.sig, which appears to be an outdated firmware compared to other users I've seen.

I'd say perhaps we could make a file pointing to the various firmware packages needed and download them manually, but I'm assuming that may cross some type of boundary with Bambu?

micahfocht, 2024-04-30:
There already is a file in the repository with a link to the latest firmware download.  I was saying that I downloaded that and put it on the SD card and it was still not working.

Alaric323, 2024-04-30:
I wasn't aware of the "latest firmware pointer", good to know that exists on some level.  However, the version the printer is requesting may be different from that latest firmware file.  Can you check to see the file name of the firmware package the printer is trying to download?  It shows on the printer screen briefly and you have to confirm it.

[image]

Alaric323, 2024-04-30:
That file is pointing to a different firmware than my printer is trying to download.  Do the bambu firmwares function on a patching system (one base firmware + extra files applied to it) or is each patch all inclusive?  Otherwise, I don't know why my printer is trying to download v00.00.28 instead of v00.00.32.

micahfocht, 2024-04-30:
If you click the "History" button at the top right of that file, you'll be able to view older versions of it.  It looks like version [`1.07.03.00`](https://github.com/X1Plus/X1Plus/blob/0435ce2d5c9f41bc03af6d94038d0192fbeab891/installer/base.json) is the version it's looking for.

Alaric323, 2024-04-30:
That appears to be correct, and since 1.1 came out before the move to 1.07.03.05, it seems that 1.07.03.00 is the intended firmware to download.  However, I've downloaded that to the SD card manually and it still doesn't pick it up.  Is it supposed to be on the root or inside the x1plus folder?

I also got a completed installer log.  It's reporting 403 Forbidden in the same way your initial issue request when trying to download the file.  Log attached.
[x1plus_installer_log.txt](https://github.com/X1Plus/X1Plus/files/15168741/x1plus_installer_log.txt)

micahfocht, 2024-04-30:
Interesting development, I logged in to the printer over SSH downloaded the firmware using wget and then ran the install.sh script.  It errored out, but after that I was able to use the installer and complete the installation.

Alaric323, 2024-04-30:
On my end, I placed the firmware manually on the SD card in the x1plus folder and it also completed installation properly.

kbhse, 2024-05-01:
Hi, I have the same issue as described here. I'd like to try the solution you found. I think you're referring to manually downloading 1.07.03.00 to the x1plus/firmware folder on the SD card. If so, where can I download 1.07.03.00 from?

Alaric323, 2024-05-01:
The link to it is in the file Micah shared earlier: https://github.com/X1Plus/X1Plus/blob/0435ce2d5c9f41bc03af6d94038d0192fbeab891/installer%2Fbase.json

Grab the URL from that json, throw it into your browser, and it'll download.

kbhse, 2024-05-01:
That's it! Thanks.

kbhse, 2024-05-01:
I have x1plus installed now :)
Thanks for your help.

jphannifan, 2024-05-01:
> The link to it is in the file Micah shared earlier: https://github.com/X1Plus/X1Plus/blob/0435ce2d5c9f41bc03af6d94038d0192fbeab891/installer%2Fbase.json
> 
> Grab the URL from that json, throw it into your browser, and it'll download.

You can also run wget on the printer and download it directly to your sd card if you want.

boomer20021, 2024-05-04:
how do you run it from the sd card

jphannifan, 2024-05-04:
> how do you run it from the sd card

You log into an SSH session on your device. Navigate to /mnt/sdcard/x1plus/firmware/

and then run 
`wget URL`

replacing URL with the URL of the update file. I'm not sure which one you need

jphannifan, 2024-05-05:
We have a fix ready for this. If anyone encounters difficulties with this update, please let me know. After installing this, your boot screen should display a wifi icon in the top right corner of the screen. You will not see it until after a successful install, though.

https://github.com/X1Plus/X1Plus/releases/download/v1.0-7-brcmfmac/1.1_rc1-33+g2d382aa.x1p

jwise, 2024-05-06:
Added instructions to https://github.com/X1Plus/X1Plus/releases/tag/x1plus%2F1.1 for where to get firmware images.  But in the mean time, Bambu have also added an exemption to their CDN configuration for the X1Plus installer, so the installer should work from the network again.  There are changes going in in the future to improve download reliability in the installer.

closed 2024-05-19 by micahfocht