X1Plus/X1Plus #240: Bug: SFTP binaries missing

jphannifan, 2024-05-06:
Build: PR 239 commit 2d382aa (01.07.03.05)

After installing MC firmware version 32.17 for the first time, the binaries for SFTP in /usr/libexec/ are now absent. When I attempt to use SFTP or a command such as `scp` in ssh, I receive this message:
```
sh: /usr/libexec/sftp-server: not found
scp: Connection closed
```

These files are indeed missing
```
[root@BL-P001-sdcard:~]# ls -l /usr/libexec/sftp-server
ls: /usr/libexec/sftp-server: No such file or directory
```

DavidJ confirmed the same finding after installing the same build. After finding this I installed a build from before the 01.07.03.05 rebase, and both sftp-server and the symlink appear in /usr/libexec/.

jphannifan, 2024-05-06:
Dropping these files into /usr/libexec/ restores SFTP instantly:

`sftp-server`
`gesftpserver`

bani6809, 2024-05-07:
bambu deliberately removed from the base fw. i guess since we do our own dropbear we can do our own sftp-server/gesftpserver?

bani6809, 2024-05-07:
fixed by 21f3de640758b16e87228d87dcc8c11445cf8932

closed 2024-05-07 by bani6809