X1Plus/X1Plus #253: reports of x1plus-settings having weird defaults

jwise, 2024-05-09:
I need to, like, blast the SD card on my printer and start fresh or something and see if I can reproduce this.

jphannifan, 2024-05-14:
Weird behavior I've been noticing lately. My printer is running the same x1p as the artifact build from PR 260 (kexec stuff)

Regardless of the setting I have for `homescreen.graph.enabled` the graph is displaying no matter what. 
[image]

I had no `homescreen.image.printing` defined and `homescreen.graph.enabled` true/false (same result either way)

I just set `homescreen.image.printing` to a .png on my SD card and now the lines appear on the plot but no background image.

jphannifan, 2024-05-14:
Update, this appears to be related to `x1plus-test-settings.py` saving everything as a string. 

When settings.json looks like this, the graph is enabled no matter what:
```
"homescreen.graph.enabled": "false"
```

However change it to this and the issue is gone:
```
"homescreen.graph.enabled": false
```

jwise, 2024-05-14:
Yeah, that's correct.  `x1plus-test-settings.py` is very much not industrial-strength right now.  On the command line if you wish to delete a key, try `x1plus-test-settings.py homescreen.graph.enabled=None`.  Once the OTA stuff is a little more stable and there are more interesting X1Plus endpoints to hit, `x1plus-test-settings.py` will become a more full-fat CLI tool.

jphannifan, 2024-05-16:
> Yeah, that's correct. `x1plus-test-settings.py` is very much not industrial-strength right now. On the command line if you wish to delete a key, try `x1plus-test-settings.py [...]

The only issue I'm still having is I'm unable to view the plots when a print is running. See the image above, that's what it looks like if I print with `homescreen.graph.enabled` set to True. It does display everything correctly while the printer is idle, so I have a feeling it's something weird with Home2Page.qml

jphannifan, 2024-06-09:
@jwise I think this issue is fixed now. Any recent reports before we close this?

jwise, 2024-06-09:
I also think this is fixed now, with the new `x1plus settings` tool that ought be quite a bit more bulletproof.

closed 2024-06-09 by jwise