X1Plus/X1Plus #279: FR: Removal of the touchscreen feature

jphannifan, 2024-05-24:
Ever since VLC Player got way faster on X1Plus, It's all I am using to control my printer now. It's to the point that I'm using my printer from VLC running on my iphone! 

I'm having a hard time seeing what the point of the dumbass screen on the printer is for? Can we get rid of this in version 2.0? I feel like our screens could be put to better use if they just weren't there?

nofuturekid, 2024-05-31:
Then just don't use it.

jwise, 2024-05-31:
I think this is a high priority issue, agreed @jphannifan

psiberfunk, 2024-05-31:
Maybe we could just entirely disable the screen output to save RAM ?  I can't imagine most X1Plus users need the screen.

cs2000, 2024-06-06:
You can control the touchscreen from VLC player?!

nofuturekid, 2024-06-08:
I do know who it is, but it has to be a joke. I mean, if YOU don't need the screen, fine, but I can't be the only one who uses this a lot. If then, make it optional to do without the screen but don't just turn it off.

jwise, 2024-06-10:
This one didn't make the cut for 2.0 but given the new features it could work for 2.1