X1Plus/X1Plus #284: Reported MQTT WiFi Signal is not correct

WolfwithSword, 2024-06-04:
**Describe the bug**
Since WiFi driver was modified in X1P 1.1, the reported signal strength from the printer's MQTT is hard-stuck at -90dBm. 

As a result, bambustudio/orcaslicer detect low signal strength (does not warn user, but evident in the signal icon) and as a result, will be more likely than not to send print requests via cloud (HTTPS-routing) rather than local (FTP), as can be viewed by monitoring mqtt data for project_file requests.

**To Reproduce**
Steps to reproduce the behavior:
1. Load X1P 1.1 with new wifi driver changes
2. Monitor MQTT push_status for wifi_signal
3. View it is stuck at -90dBm
4. Optional: Start a print and monitor /request mqtt topic and view it is from cloud.

**Expected behavior**
WiFi signal from MQTT should reflect the proper signal strength. And if it is high enough, printing should prefer to use local FTP instead of cloud service & slicer icon for signal strength will be higher.

**Screenshots/Photos**
[image]

[image]
User in second screenshot had just installed X1P, so their -30dBm value before was pre-X1P 1.1

**External Links**
Has this bug been discussed elsewhere that you can link to? Have you checked this repository's discussions section?

https://discord.com/channels/1041781310299062282/1233015570610524160/1247333181745659996 

**Additional Details:**
 - Printer type (X1/X1C/X1E): X1C
 - X1Plus version: 1.1
 - AP firmware version:
 - AP hardware revision:
 - MC firmware version:
 - MC hardware revision:
 - Toolhead firmware version:
 - Toolhead hardware revision:
 - Computer vision model:

**Is the bug present on OEM firmware?**
Did not test

**Custom modifications**
Additionally monitored via NodeRed to view wifi signal

**Additional context**
It is reported that checking `iw dev wlan0 station dump` reports the correct value.

Copied from @jwise in discord:

```
looks like forward tries to open("/proc/net/wireless")
indeed, in bbl_utils there is a function 493: 000671f8  2108 FUNC    GLOBAL DEFAULT   11 wifi_link_quality[abi:cxx11]()
so probably overriding that is how to do it
```

jwise, 2024-06-07:
Reported fixed by #285 .

WolfwithSword, 2024-06-07:
Tested build with that fix - can confirm the signal strength reported in MQTT is now accurate and updates frequently. Also reports as higher in the slicer now.

[image]
[image]
[image]

closed 2024-06-07 by WolfwithSword