X1Plus/X1Plus #309: <your bug name here>

hainjedaf, 2024-06-16:
**Describe the bug**
During install, installer locks up trying to download firmware.
Wifi connection is lost.
After that, sd card is no longer bootable.

**To Reproduce**
1. power down printer
2. remove SD card
3. put x1plus 2.0_RC0 x1p file on SDcard root
4. insert SD card
5. power on printer
6. chose 'Start x1plus install'
7. chose 2.0_RC0
8. wait for installer to unpack and start install process
9. printer reports wifi disconnect and gets stuck on downloading firmware.
10. reboot or powercyle printer
11. go back to step 1

**Expected behavior**
Full install of X1C 2.0_RC0

**Screenshots/Photos**
If applicable, add screenshots or photos to help explain your problem.

**External Links**
Has this bug been discussed elsewhere that you can link to? Have you checked this repository's discussions section?

**Additional Details:**
 - Printer type (X1/X1C/X1E): X1C
 - X1Plus version: 1.1
 - AP firmware version: 
 - AP hardware revision:
 - MC firmware version:
 - MC hardware revision:
 - Toolhead firmware version:
 - Toolhead hardware revision:
 - Computer vision model:

**Is the bug present on OEM firmware?**
Have you tested whether this bug is also present in OEM firmware?
If it is, which version? *ex. 01.07.02.00*
Are any steps to replicate it different?

**Custom modifications**
Add details of any software or electronic modifications you've made to your printer here.

**Additional context**
Add any other context about the problem here.

josh5137, 2024-06-16:
I am having the same exact problem.  It connects to wifi fine on oem and when x1 plus is fully loaded.  it just doesnt work on the installer so it fails updating.  I dont have any custom modifications or anything.

jwise, 2024-06-17:
Thanks for this report.  @josh5137 and @hainjedaf , could you please provide a copy of the `x1plus_installer_log.txt` from your SD card?  We've heard a few reports of this and we'd like to track it down the best we can.

Thanks!

hainjedaf, 2024-06-17:
I added my x1plus installer log...

Reportedly Bambu Labs have changed the CDN for their firmware updates.
Hence the current 2.0 install looks in the wrong place.

For the time being, the following workaround worked for me. Thanks to user Jon on X1Plus discord.

[WORKAROUND]

Download the following: 

https://public-cdn.bblmw.com/upgrade/device/BL-P001/01.08.00.00/product/45ea644d25/update-v00.00.32.18-20240422102213_product.img.zip.sig

Place this in x1plus/firmware on the SD card

Place 2.0_rc0.x1p in the SD card root. 

This can be done via FTP or by switching off the printer and transferring the SD card to your PC.

Once printer boots, start the update process as normal. the updater will find then .img.zip.sig file and proceeed as normal

After that, run the OTA updater to upgrade to 2.0-rc1 and test the OTA functionality as described.

[end WORKAROUND]

[x1plus_installer_log.txt](https://github.com/user-attachments/files/15865603/x1plus_installer_log.txt)

jwise, 2024-06-17:
OK I took a look at this log and this is *really* interesting.  It looks like everything in the wpa_supplicant.conf is disabled, but given that you say it works in a real booted environment, I think something else must be getting used there.  I did some *more* looking and it looks like `netService`, these days, starts up with `wpa_supplicant_new.conf` -- but `wpa_supplicant_hook.sh` starts up with `wpa_supplicant.conf`.  I think the solution here is that instead of starting up `wpa_supplicant_hook`, we should start up `netService`.

This is a little difficult to fix because `kexec_ui` starts up `wpa_supplicant_hook`, not `install.py` (and `kexec_ui` is not overwritten as part of the 2.0 install process).  But we can have `install.py` detect if `netService` is not running, shut down the "bad" `wpa_supplicant`, and start it up via `netService`.

Thanks for the installer log!  This definitely points to a fix.  I'll come back in a few days with something for you to try, if you don't mind.

josh5137, 2024-06-17:
Thank you hainjedaf.   That workaround worked for me too.  I was able to complete the 2.0 update and the OTA update now.

hainjedaf, 2024-06-18:
> OK I took a look at this log and this is _really_ interesting. I

Glad to be of help.

> This is a little difficult to fix because `kexec_ui` starts up `wpa_supplicant_hook`, not `install.py` (and `kexec_ui` is not overwritten as part of the 2.0 install process). But we can have [...]

Does it manage to capture the WIFI credentials?
Also: I run my X1C on an IoT network which has hidden SSID and allows only known MAC addresses.
Changing the Wifi to SSID vissible and any MAC allowed, did not help.
 
> Thanks for the installer log! This definitely points to a fix. I'll come back in a few days with something for you to try, if you don't mind.

Again, glad to have been of help. It was user Jon on X1Plus discord who got me the workaround.

I am also capable of downgrading my printer to Bambu firmware 1.06 to do complete reïnstall of X1Plus if needed.

Kind Regards,
Marout Sluijter-Borms
a.k.a. HainjeDAF
Amsterdam NL

closed 2024-06-19 by jwise

hainjedaf, 2024-06-19:
@jwise Will I still get some things to test?

jwise, 2024-06-19:
@hainjedaf If you wouldn't mind, could you try:

1) Grab the latest build from CI here: https://github.com/X1Plus/X1Plus/actions/runs/9587053339 (download the x1p artifact, unzip it, put the unzipped artifact on your SD card).
2) Delete /x1plus/firmware/update-v00.00.32.18-20240422102213_product.img.zip.sig and /x1plus/images/00.00.32.18.squashfs from your SD card (if you're doing this ssh'ed into your printer, rather than rm'ing 00.00.32.18.squashfs, just mv it to a different filename).
3) Reboot and run the new installer, and see if it succeeds at downloading this time.

Thanks,
joshua

hainjedaf, 2024-06-19:
Tried it. 
Crashes on network connection. "Network is unreacheable."

I include the installer log.

[x1plus_installer_log.txt](https://github.com/user-attachments/files/15906565/x1plus_installer_log.txt)

jwise, 2024-06-19:
@hainjedaf Oof.  Thanks for the diagnostics on that one. Ok, one more thing to try.  If you boot the printer into OEM firmware, does the WiFi automatically connect?  (You can do this by holding down both buttons on the top of the printer while powering it on; you can release them when the chamber light turns on.)  If it doesn't, what happens if you connect to WiFi there, and then run the latest build with these steps again?

@josh5137 When you said "it connects to wifi on OEM", did you mean "before you installed X1Plus, it connected to WiFi fine", or did you mean "after booting into OEM firmware while this issue was happening, it connected to WiFi"?

hainjedaf, 2024-06-20:
@jwise No it did not...
Maybe I did not start off with an OEM config that could.... I'm not sure...
I upgraded from X1Plus 1.1 to 2.0-rc0

So I re-entered the credentials in OEM mode and tried again.
OEM connects (hidden SSID, MAC access restricted)
However,
-OEM reports incosistencies in firmware (1.06.)
Reboot:
-OEM does not retain WIFI credentials over reboot.

SO I re-enter credentials. Let OEM reconnect.
Power down and repeat your instructions on x1p, firmware and images...

This time it WORKS!

[x1plus_installer_log.txt](https://github.com/user-attachments/files/15906928/x1plus_installer_log.txt)

jwise, 2024-06-20:
Cool -- thanks for the update!  I'm not sure if there's going to be a better fix forthcoming for 2.0, but this definitely seems like the sort of thing that ought go into the release notes.  It also should be mostly obviated by the OTA download system.

hainjedaf, 2024-06-20:
Well, I think it will only be relevant for the instances where you move
from stock firmware to x1plus.

PR #313 (2024-06-19): use netService in installer instead of wpa_supplicant_hook
There have been some reports that WiFi works on OEM and works in X1Plus, but not in the installer -- e.g., #309.  In modern OEM firmware (and, indeed, in X1Plus!), the network is managed by `/usr/bin/netService`, rather than by `wpa_supplicant_hook`.  And, in fact, `wpa_supplicant_hook` sometimes can use the wrong config file (though, oddly, in Firmware R, `netService` does not use the "wpa_supplicant_new.conf" like it does in the X1Plus base firmware, but all the same, the user in #309 reports that WiFi works booted into OEM, so ...). [... cut]
commit 306f9446 (2024-06-19): use netService in installer instead of wpa_supplicant_hook (#313)

There have been some reports that WiFi works on OEM and works in X1Plus, but not in the installer -- e.g., #309.  In modern OEM firmware (and, indeed, in X1Plus!), the network is managed by `/usr/bin/netService`, rather than by `wpa_supplicant_hook`.  And, in fact, `wpa_supplicant_hook` sometimes can use the wrong config file (though, oddly, in Firmware R, `netService` does not use the "wpa_supplicant_new.conf" like it does in the X1Plus base firmware, but all the same, the user in #309 reports that WiFi works booted into OEM,  [... cut]
commit 6f53f99b (2024-06-28): use netService in installer instead of wpa_supplicant_hook (#313)

There have been some reports that WiFi works on OEM and works in X1Plus, but not in the installer -- e.g., #309.  In modern OEM firmware (and, indeed, in X1Plus!), the network is managed by `/usr/bin/netService`, rather than by `wpa_supplicant_hook`.  And, in fact, `wpa_supplicant_hook` sometimes can use the wrong config file (though, oddly, in Firmware R, `netService` does not use the "wpa_supplicant_new.conf" like it does in the X1Plus base firmware, but all the same, the user in #309 reports that WiFi works booted into OEM,  [... cut]