X1Plus/X1Plus #319: Show how much filament is left on each spool (AMS) -> without BambuLab RFID Taged Spools

TronickDev, 2024-06-21:
**Is your feature request related to a problem? Please describe.**
No, it is not related to a problem.

**Describe the solution you'd like**
With the printer, I got a Green BambuLab filament. After it was empty, the spool was replaced with a SUNLU black spool. The AMS did not show how much is left on this spool, even after I selected the type and color that the AMS shows the slot correctly. Then I just put the RFID tag from the empty Green BambuLab filament spool on the SUNLU spool, and then the filament was quite correctly recognized. Sure, the AMS now says that it is green PLA and not black PLA, but how much was left was shown correctly. I also tried the refill option with this black PLA spool, and this was also working.

In this case, I think that the RFID tags hold information about the width, inner and outer diameter of the spool, as well as how much was on it initially (250g, 500g, 1000g).

So may it be possible if we provide the information by hand, and then it should be possible to show how much is left on a spool without needing an RFID tag?

If so, you just need to ask for that information on the same screen, where you need to select the type of filament and the color. 

Maybe just adding some fields to this screen in order to type in the missing information, or adding spools to the printer and then choosing the preset, and then selecting a preset for the AMS slot. This would make it easy to select.

I will look into this, but since I'm new to the X1Plus firmware, I need to look a bit at how things work here.

**Describe alternatives you've considered**
Creating own RFID Tag correctly signed that the BambuLab software can handle it.

**Software Information**
 - Current X1Plus Version: BambuLab X1C with X1Plus 2.0 rc1
 

**Additional context**
none

jphannifan, 2024-06-21:
The filament usage data reported on the touchscreen and in the slicer is generated mostly from data from the RFID readers. 

The AMS pulls a predefined length of filament from each tray, and it counts how many rotations each RFID tag makes. Using # of rotations and the data from the RFID tag like density and initial weight, it can accurately estimate the weight remaining.

In order to pull the same length of filament from each tray every time, there's an odometer in the filament hub of the AMS that measures length, so it's using this length data combined with spool rotations.

See the wiki for more info - [https://wiki.bambulab.com/en/ams/manual/ams-function-introduction](https://wiki.bambulab.com/en/ams/manual/ams-function-introduction)

The first step (in my thought process, anyway) is to determine if we can use the odometry data. We don't have direct access to this sensor, so we're limited to passively monitoring log files to record length data that prints every few seconds. I guess I want to see if it's possible to do custom RFID tags without reverse engineering the AMS firmware. You can get 2 RFID readers for < $10 so maybe it's possible to install new readers alongside the old ones? With I2C adapters, we could control the readers from X1Plus, or I guess that could be done with an esp32

Roving-Ronin, 2025-01-20:
Keep an eye upon https://openspool.io/ mention on there of a version for the AMS, that would allow using USD$0.20 tags to ID non Bambu spools (and still read Bambu spools also)