X1Plus/X1Plus #322: 3 points bed tramming

Alkerion, 2024-06-24:
**Describe the solution you'd like**
A clear and concise description of what you want to happen. Ex. "Change menu item to be z instead"
When I'm on the manual bed tramming calibration menu I'd like to be able to perform a quick three points tramming.

I would like to have a button to test the bed distance for each point, that will allow to quickly level the bed without to have to run 30 times the full diagnostic.

**Software Information**
 - Current X1Plus Version: 2.0rc

Alkerion, 2024-06-27:
Yes, that's more or less my request.

Been able to perform a quick test first (center then 3 points above knobs), then been able to test each point separately will be nice to have.

Of course the Z value of each point should be reported.

It will allow for a quick tramming before fine tuning with the full test if required.

More precise and useful than with a sheet of paper IMHO.

kylek29, 2024-10-21:
So I noticed the Z-info is displayed in 2.0+ for the bed calibration procedure (for each probe point) as a popup. I think if we can get that same mechanism in the tram section it would fulfill this and really help with the fact that BambuLab doesn't give us any of these coordinates on the machine (there's been many times I've had to resort to manual calculation).