X1Plus/X1Plus #327: installer for mac doesn't work

StephInoP, 2024-06-27:
**Describe the bug**
installer for mac doesn't work

[image]

psiberfunk, 2024-06-27:
Odd , I tested this on my m3 Mac and worked ok.  What architecture are you on ?

psiberfunk, 2024-06-27:
I suspect the problem here is the "x64" build is actually mismarked and the bug reporter is running an intel mac and we provided an arm-only build, based on inspection on my mac, I see this.. which would say universal (arm+intel) or intel if it was runnable on intel macs:

[image]

StephInoP, 2024-06-27:
> Odd , I tested this on my m3 Mac and worked ok. What architecture are you on ?

I can open v1.1, same laptop same OS

[image]

psiberfunk, 2024-06-28:
Ok , so you're on intel.    This is probably because in 1.1 we built an intel-only installer.  Something changed here where we build a arm-only installer.. which obviously doesn't run on intel (but vice-versa is fine).  See this is the get-info on the 1.1 installer:

[image]

jwise, 2024-06-28:
Yeah I bet I know how I f’ed this up. I’ll fix it when I get home.

Qnapper, 2024-08-24:
Still cannot use unstaller as Macbook running OS15 does not allow it. Please help

jphannifan, 2024-08-31:
> Still cannot use [X1Plus 2.0](https://github.com/X1Plus/X1Plus/releases/tag/x1plus%2F2.0) installer as Macbook running OS15 does not allow it. Please help jwise said he would fix it on 28 Jun. Strange thing is I can open X1Plus 1.0 no problem!!!!!

I am running Sonoma and it works fine for me. Could you provide more details like the exact error message you get (screenshot is fine too). I have not tried Sequoia yet but I have no trouble on macOS 14

shitwolfymakes, 2024-09-09:
Same issue here. I'm running MacOS Monterey 12.7.5

Qnapper, 2024-09-09:
Anyone Please would love to get this installed. Thanks

jwise, 2024-09-09:
OK I am a dum dum and what happened is I built for arm64 instead of x86_64.

The workaround for now is to use the X1Plus 1.0 installer, and before you do, copy the 2.0 .x1p file to your SD card and select it at install time.

Ugh, sorry about this, y'all.  jwise did say he'd fix it and he didn't because he is bad and keeps getting distracted.  But this workaround should help.

shitwolfymakes, 2024-09-09:
@jwise is there a CI/CD action that builds the installer? I don't see anything that would do that in the current workflows

jwise, 2024-09-09:
@shitwolfymakes No, there isn't.  This is part of the "@jwise is bad" problem.  The part I got stuck on for CI/CD is 1) getting CI/CD set up to be able to build and code-sign a Mac binary, and 2) figuring out how to get my Apple developer keys into CI/CD in a way where I can personally oversee their usage.

I would be very appreciative if you had hints for either of these!

jphannifan, 2024-09-14:
Since the CI/CD discussion has continued on Discord, I'm going to go ahead and close this issue since Joshua has fixed this one

jwise, 2024-09-14:
no, this is still open, it is not fixed

FKentD, 2024-09-20:
Got the same result on my 2017 iMac. That's when I opened the file's Get Info and seen "For Apple Silicon" so I'll try again tomorrow on my M2 MacBook Air. Intel support for Macs may soon die out for smaller organizations like this due simply to cost of development. just my opinion.

uberjay, 2024-11-04:
Still no Intel installer -- probably the best option for me is to follow the offline installation instructions?

tsndr, 2024-12-17:
I want to install X1Plus but I only have an Intel Mac, so I can't until this is fixed :/

rwbot, 2024-12-25:
> @shitwolfymakes No, there isn't. This is part of the "@jwise is bad" problem. The part I got stuck on for CI/CD is 1) getting CI/CD set up to be able to build and code-sign a Mac binary, and 2) [...]

@jwise any chance you could make a quick command list of setting the installer build to target intel? gh workflow would be perfect but even the flow for locally compiling it from src would be awesome too

Karlosek, 2025-01-25:
> The workaround for now is to use the X1Plus 1.0 installer, and before you do, copy the 2.0 .x1p file to your SD card and select it at install time.
> 

Worked for me.

jwise, 2025-06-17:
Nobody has said it doesn't work, so I guess I declare it works.

closed 2025-06-17 by jwise