X1Plus/X1Plus #336: Chamber fan howling

G4Cab, 2024-06-30:
Why is the chamber fan speed so bad regulated in it's hystereses?

It's damn annoying switching every second from 80% to 90% to 80% to 100% to 90%......

Argh, i'll go crazy!

ccatlett1984, 2024-06-30:
If you boot into stock firmware, does the issue persist?

G4Cab, 2024-06-30:
> If you boot into stock firmware, does the issue persist?

Hmm, how should i test it?
Reboot to original fw won't take me to newest original fw?

Or should i upgradey test, and go back?

I think it's a new feature, than i never seen the chamber fan regulating it's speed automatically to the chamber temp.

Hawk30100, 2024-07-02:
There is indeed a built-in feature on the stock X1C firmware which regulates the chamber fan, aux fan and nozzle temp, when the chamber temp reaches the softening temperature of the filament set in the AMS.

I did several tests and could watch how my printer slowly increased the chamber fan on its own, when the chamber temp reached about 38 degrees Celsius, when printing PLA with AMS. The fan went up to 100% at about 42-43 degrees Celsius. AUX fan to 70% at about 44-46 degrees and nozzle -5 degrees Celsius.

(I do print with top glass and door closed. Sometimes with ambient temp  up to 32 degrees Celsius in summer)

jphannifan, 2024-08-30:
The only way X1Plus would be able to change fan speed is by publishing gcode or by changing a specific property in QML. As of the most recent commit, there is no code in our repo that I can point to as being a source of this issue.

closed 2024-08-31 by jphannifan