X1Plus/X1Plus #338: Display the real print duration

Alkerion, 2024-07-02:
**Is your feature request related to a problem? Please describe.**
The real print duration isn't displayed when the print is complete, only the calculated one from the slicer, who's most of the time rather inaccurate.

**Describe the solution you'd like**
Display the real print duration instead of the calculated one.

**Describe alternatives you've considered**
None

**Software Information**
 - Current X1Plus Version: 2.0

**Additional context**
[image]

jphannifan, 2024-07-04:
I don't think we want to be replacing the slicer estimate, but maybe we could display the actual print time above the slicer estimate, with labels of course. 

For now a more realistic goal is to start logging and displaying the actual print time next to the estimate before trying to correct the estimated value. In the meantime, I would check to see what your slicer settings are for 'filament load time' and 'filament unload time' under "Basic Information" in device settings (Bambu Studio or Orca) 

Bambu says that the estimate's accuracy depends on slicer parameters: https://github.com/bambulab/BambuStudio/issues/703

Alkerion, 2024-07-04:
For me, when the print is complete I don't really care about the estimated time anymore, but only the real print time.
It's like the weight, I don't really care about it on the Complete! print dialog screen, I don't see any interest to have it displayed.

I know how to set the "filament (un)load time", but it's not the Graal, I've already tried to change them a few times based on real print times (without troubles during print), but not always close to reality.