X1Plus/X1Plus #339: Display the users names for LAN access

Alkerion, 2024-07-02:
**Is your feature request related to a problem? Please describe.**
The users names required to connect to the printer aren't displayed, and nothing is found on the user guide too.

**Describe the solution you'd like**
Display the users on the LAN access screen, and add logging information on the user guide.

**Describe alternatives you've considered**
None

**Software Information**
 - Current X1Plus Version: 2.0

[image]

psiberfunk, 2024-07-02:
I am extremely confused. That screenshot shows the user

Alkerion, 2024-07-02:
> I am extremely confused. That screenshot shows the user? Is it a mock ?

Yes, it's a mock up I quickly PS to show what I expect.

psiberfunk, 2024-07-02:
So, I think the root user password doesn’t really need to username because the username is obvious in the word “Root”.

the bblp isn’t really meant to be logged into via ssh as far as I can tell, and it would likely end up misleading users to try and ssh into something they don’t want … so why show them a door to the wrong place ?

Alkerion, 2024-07-02:
It's also used to connect via FTP, so it's mandatory to have both user and pass to connect.

Everything is obvious when you know it.

At the moment I found the user info nowhere easily, reason why I suggest to indicate it on the screen and/or the user guide.

jwise, 2024-07-02:
I agree that 'root password' really does imply that the username is `root` (and, users who do not understand that the root password goes to a user named `root` probably should not be SSHing to their printer as root for *any* reason!).

I also agree it would be good to have a section that says "MQTT and FTP+SSL are always enabled.  Use the username "bblp" and the LAN access code as credentials.", or something like that but with better wording.

closed 2024-09-30 by jphannifan