X1Plus/X1Plus #344: Web UI features

jphannifan, 2024-07-11:
These are my initial thoughts on what kind of functionality the web UI should provide - at least to get a discussion about it started. We have had tons of feature requests that overlap with this one in a broader sense (e.g. scheduling LEDs), and those we will review in the near future. 

All users are encouraged to provide any input that might be useful here. Please note though that this is not meant to be an exhaustive list of feature requests but more of an overview of the features or sets of features we want to focus on the most. We have web VNC, so I think the last thing we need is a web interface that is just a carbon copy of the touchscreen interface.

Here is what the Octoprint landing page looks like, and maybe there is some inspiration we can take from it
[image]

These are the key features that stand out as things we should implement:
- Print queue and print progress
- Print history
- Usage statistics (this one might be a future goal)
- Gcode terminal
- Calibration - vibration comp, bed mesh, tramming
- Camera stream/camera preview
- Macros/add-ons *

* We removed macro features from X1Plus 1.0 because it just wasn't implemented the right way. The web UI is a perfect place to revive this feature in my opinion. I think we should avoid referring to it as a "plugin" platform, but this could work similarly to [Octoprint plugins](https://plugins.octoprint.org/) which are all written in Python. 

Since many users want to use Gcode macros but are not interested in writing a Python macro, maybe there should be two separate features, one being "Gcode macros" and the other "Python add-ons".

jneilliii, 2024-07-11:
I think one of the must haves is a file list that can be used to start prints from files previously sent to the printer. I guess that is kind of implied based on wanting print queues.

zippytiff, 2024-12-09:
This would be high on my list, unless somehow the X1Plus features can show on the BambuLabs app ?

So ideally having an X1Plus webpage, it can develop, as the features develop

Thoughts ?