X1Plus/X1Plus #349: Ability to change flow ratio by X% mid-print.

ctbrahmstedt, 2024-07-21:
**Describe the solution you'd like**
Like every other 3D printer in existence, add the ability to change the flow ratio by a percent while printing. It would be great if it could be independent of each filament. ie. While printing on AMS Color 1, I can change ratio to 95%, but when it changes to AMS Color 2, it stays at 100%, or rather can be set independently, and when it switches back to AMS Color 1, it remembers the flow ratio. 

I guess maybe this interface in the AMS manager, where you can control each filament, a setting for each filament that can be adjusted mid-print. With maybe an option of, "Maintain settings after print?" 

I can't believe Bambu doesn't have this as an option, but definitely hugz n kisses to the team if this feature can be enabled. 

Thanks!

jneilliii, 2024-07-21:
Already in the works....

https://github.com/X1Plus/X1Plus/pull/343

kylek29, 2024-07-22:
For what it's worth, I think there is a BambuLab GCode for this if you wanted to trigger it via the slicer (say, on a filament change or from some logic).

For example, this might work:

Silent
M220 S50

Sport
M220 S124

Ludicrous
M220 S166

jphannifan, 2024-07-28:
M220 is for adjusting the feed rate. To adjust the speed properly, you need to use M204.2 as well. You can view the function I wrote for it here to see what the full set of commands looks like. If you exclude M204.2, acceleration is not adjusted, and your extrusion rate will not be consistent. https://github.com/X1Plus/X1Plus/blob/253f47758d6ac6d6d7d6df9c68d46e4ff5a791de/bbl_screen-patch/patches/printerui/qml/x1plus/GcodeGenerator.js#L546-L549

The command you want to use for flow ratio is M221, ie `M221 S99.5`