X1Plus/X1Plus #354: AMS and AMS hub updates not possible

Alkerion, 2024-08-03:
**Describe the bug**
Can't update AMS and AMS hub to latest version, the update menu told me I've the latest versions, that's wrong as newest versions exists.

**To Reproduce**
Steps to reproduce the behavior:
1. go to "General tab" under Options
2. Tapon "Firmware version"
3. Tap on "AMS hub" or "AMS #x"
4. Update not available

**Expected behavior**
Been able to update AMS and AMS hub

**Additional Details:**
 - Printer type (X1/X1C/X1E): X1C
 - X1Plus version: 2.0

AceAsket, 2024-08-04:
same problem

Alkerion, 2024-08-13:
Any chance to have an answer regarding this issue ?

jwise, 2024-08-13:
What version do you expect to be able to update to, and what version are you currently running?

Alkerion, 2024-08-14:
I expect to update to latest FW relases from Bambu :
https://wiki.bambulab.com/en/x1/manual/X1-X1C-AMS-firmware-release-history

My Fw are :
[image]

And I've these msg when I want to update :
[image]
[image]

ScottJD, 2024-08-26:
I have not installed Plus yet. But form what I understand you can power down, remove the SD card and power back up to stock firmware and should be able to update the AMS like that. Or if you don't have/allow internet connection to the printer you can follow the OTA offline SD card update procedure form Bambu Labs to update the AMS by putting the AMS firmware on a different SD card and update the AMS that way. I am not saying it’s not a bug, but for those that find this important for the RFID’s of new filaments this is a work around for now. 
https://wiki.bambulab.com/en/x1/manual/X1-firmware-update-from-SD-card

lightmaster, 2024-09-10:
> I have not installed Plus yet. But form what I understand you can power down, remove the SD card and power back up to stock firmware and should be able to update the AMS like that. Or if you don't [...]

Not possible. When booting back to the stock firmware, it won't let you do anything without first updating the stock firmware to the current version, breaking root. And you can't do an offline update since that was only introduced in 01.08.02.00 and the rootable firmware is based on an older version which I can't remember the number of.

I suppose you could update the stock firmware and AMS, unplug the AMS, install Firmware R, then install X1Plus again, then plug the AMS back in again.

jwise, 2024-09-10:
If you want to experimentally mess with this, you can edit the `/opt/x1plus/etc/version.json` file to upgrade the AMS.  There is a filament.json upgrade sequence that I don't really understand right now, but you probably can experimentally upgrade the AMS given the new ota.json from an upgrade.

jphannifan, 2024-09-11:
> If you want to experimentally mess with this, you can edit the `/opt/x1plus/etc/version.json` file to upgrade the AMS. There is a filament.json upgrade sequence that I don't really understand right now, but you probably can experimentally upgrade the AMS given the new ota.json from an upgrade.

I can try this on my printer and I'll report back here

Alkerion, 2024-09-11:
I finally connected my AMS and AMS hub to a friend P1 to update them.
Hope that we will be able to perform such update directly from our X1 printers in the future.

jphannifan, 2024-09-11:
I was able to update my AMS from 00.00.06.40 to 00.00.06.44 with an MQTT command. First you need to check to see if you need `rev7` or `rev8` firmware, then publish this MQTT command to `device/{serial}/upgrade` 

```
{
  "json": {
    "upgrade": {
      "command": "start",
      "sequence_id": "0",
      "module": "ams",
      "version": "00.00.06.44",
      "url": "http://127.0.0.1/ams_rev8-firmware-v00.00.06.44-20240703093110_product.bin.sig"
    }
  }
}
```

if you need rev7 then use this filename: `ams_rev7-firmware-v00.00.06.44-20240703093253_product.bin.sig`

Unless you've downloaded the file ahead of time and placed it where the updater expects it, you will probably receive an "Update failed" message the first time around

jphannifan, 2024-09-11:
For those who do successfully update their AMS, please ignore this dialog and the warning. We will get this fixed in the next X1Plus update. 
[image]

lightmaster, 2024-09-11:
> Then publish this MQTT command to `device/{serial}/upgrade`

I assume that's the serial of the printer, not the AMS? I tried both serials and I'm not seeing any messages on the MQTT broker about it nor on the screen.

jphannifan, 2024-09-11:
> > Then publish this MQTT command to `device/{serial}/upgrade`
> 
> I assume that's the serial of the printer, not the AMS? I tried both serials and I'm not seeing any messages on the MQTT broker about it nor on the screen.

Your device's serial number. I made a mistake when typing that out, and it should have read `device/{serial}/request`. Sorry about that. If you're continuing to receive errors, let me know. I can also share a shell script later which might be the simplest option

jphannifan, 2024-09-11:
I'll need to look through my log files to understand what the issue is here, but in the meantime try publishing this command to the same topic as before. 

```
{
    "upgrade": {
      "command": "start",
      "sequence_id": "0",
      "module": "ams",
      "version": "00.00.06.44",
      "url": "http://127.0.0.1/ams_rev8-firmware-v00.00.06.44-20240703093110_product.bin.sig"
    }
  }
```
If this one works let me know and I'll update my instructions above for everyone else

lightmaster, 2024-09-11:
Now I've got an updating prompt that's progressing

lightmaster, 2024-09-11:
I was also able to get the firmware page to stop asking me to install the previous version by change the last bit of it to:

```
"ams": {
                "version": "00.00.06.44",
                "paths": {
                    "ams07": {
                        "url": "http://127.0.0.1/ams_rev7-firmware-v00.00.06.44-20240703093253_product.bin.sig",
                        "md5": "9f52c896f0a349fc054290315db63e20"
                    },
                    "ams08": {
                        "url": "http://127.0.0.1/ams_rev8-firmware-v00.00.06.44-20240703093110_product.bin.sig",
                        "md5": "9be1ccccd3ee5e0e63ef9214b322ad5e"
                    }
                }
            }
```

That might also get it to prompt you to update instead of using MQTT.

jphannifan, 2024-09-11:
@lightmaster awesome thanks for confirming that worked! And your idea there is much better than mine, and this should be doable without messing with MQTT. I will share a simplified solution shortly.. I do not own an AMS hub, so if anyone is having issues with updating the hub, I may need some troubleshooting assistance but I'll do what I can to help.

jphannifan, 2024-09-11:
> > @lightmaster awesome thanks for confirming that worked! And your idea there is much better than mine, and this should be doable without messing with MQTT. I will share a simplified solution shortly.. I do not own an AMS hub, so if anyone is having issues with updating the hub, I may need some troubleshooting assistance but I'll do what I can to help.
> 
> I have now also updated my 2nd AMS. Simply deleted both and connected the 2nd as the only one.
> 
> If there is an update for the AMS Hub, I would need that too. So I would help if possible. But right now I am printing.

I was going to suggest this (connecting one at a time and updating them one by one) and I'm glad to hear this worked. I'm hoping we can find a solution that doesn't require workarounds, but I only have one AMS, so I'm unable to test anything involving the hub or multiple AMS'. 

Let me check which firmware file corresponds to the AMS hub, but the update process should be very similar. As @lightmaster noted though, you should be able to update by editing `/opt/x1plus/etc/version.json` and updating as usual in the UI. I want to test it real quick before I share any instructions

jphannifan, 2024-09-11:
Here is a solution that will work for anyone with one AMS. 
[ams_update.sh.txt](https://github.com/user-attachments/files/16967518/ams_update.sh.txt)

Save this shell script to your sd card and run it (`source /sdcard/ams_update.sh`) and you should be able to update now. Alternatively you can edit `/opt/x1plus/etc/version.json` manually. Refer to the shell script for the values you need to update it with. 

For users with an AMS hub, the binary for the hub update is `ahb_rev1-firmware-v00.00.00.43-20240511112057.bin`. If someone who has multiple AMS' and a hub can share their `version.json` file, that would be a big help

lightmaster, 2024-09-11:
> Here is a solution that will work for anyone with one AMS.
> [ams_update.sh.txt](https://github.com/user-attachments/files/16967518/ams_update.sh.txt)

Shouldn't that create a backup of the file first, just in case? 

> If there is an update for the AMS Hub, I would need that too.

The current update for the hub in the latest update has a datetime back in May, so your hub should already be up to date. I don't have one to verify that though.

jphannifan, 2024-09-11:
Good idea. I updated the script it to do just that. 

The latest version of the hub appears to have been released in July (v00.00.00.43) but I have never seen or used one so I'm not sure if it's necessary. What I would like to determine though is how `version.json` is organized when more than 1 AMS exists

lightmaster, 2024-09-11:
@Shadow27374 You've got both AMS units and your hub attached right? Looks like all AMSs reference the same AMS update which makes sense. I don't see the ahb in your file at all though, which is weird.

You might be able to create a new section at the end to and see if that gets an update prompt: [ams-abh-update.sh.txt](https://github.com/user-attachments/files/16969720/ams-abh-update.sh.txt)

jphannifan, 2024-09-11:
@Shadow27374 Thanks for sharing. I think in the case of multiple AMS, we'll have to use MQTT until we can get an X1Plus update ready to fix this. 

The updater is not able to distinguish multiple AMS units when it starts an update. This appears to be a matter of defining a property to specify the `name` field ie 
`"name": "ams/0"` for AMS 0
`"name":"ams/1"` for AMS 1 .. and so on.

This can be done with MQTT , but I'll need to do a little checking on what the payload needs to contain for this

lightmaster, 2024-09-11:
> I am now being offered an update to 43. Unfortunately, it doesn't do it when I click on it.

I assume you have the `ahb_rev1-firmware-v00.00.00.43-20240511112057.bin` file on your sd card? Beyond that, idk.

jphannifan, 2024-09-11:
Updating the hub will not have any effect on the filament list. There is only one release note for the hub firmware in the last few months, so it doesn't appear to do very much in the first place. 

The hub is module `ahb` so you can try to update it with an MQTT command like this:
```
{
    "upgrade": {
      "command": "start",
      "sequence_id": "0",
      "module": "ahb",
      "version": "00.00.00.43",
      "url": "file:///sdcard/ahb_rev1-firmware-v00.00.00.43-20240511112057.bin"
    }
  }
```

However this was a wild guess and I won't be able to test this for a while

jphannifan, 2024-09-11:
Thanks for confirming that. This just leaves the filament list. If you boot into OEM fw ("Start built in firmware unmodified"), it will download the newest filament file and the list is correctly populated. It doesn't automatically work in X1Plus afterward, but I think I'm close to finding a solution

lightmaster, 2024-09-12:
> I have also just noticed that all the newer filament types are still missing when selecting the external spool.

What filaments aren't reading if you don't mind me asking? I've got a decent batch of filaments coming Friday, so I might be in the same boat as you then.

jphannifan, 2024-09-12:
I just successfully updated my filament list, and to do so I had to copy the new `ota-filament*.zip.sig` file to `/userdata/upgrade/filament/`, remove any other `ota-filament*.zip` files, and reboot. 

I need to do a little more testing before I have a solution I can recommend, but for anyone who would like to test it out in the meantime these shell commands can be run on the printer and will download the necessary file to `/userdata/upgrade/filament/`
```
curl  https://public-cdn.bblmw.com/upgrade/device/BL-P001/filament/product/ota-filament-v01.09.00.22-20240902153732.zip.sig -o /tmp/tmp.zip.sig
mv /tmp/tmp.zip.sig /userdata/upgrade/filament/ota-filament-v01.09.00.22-20240902153732.zip.sig
```

Then try restarting the screen service and/or rebooting to get the list to refresh.

lightmaster, 2024-09-12:
> With the external spool, no NFC chips are read out, you have to select which filament is in this external spool yourself via the menu. All newer filaments are missing from the selection menu.
> 
> For example, PETG-HF and ABS-GF are missing.

oh, I thought you meant that you had some filament that you had ordered that wasn't being automatically recognized by the AMS.

lightmaster, 2024-09-12:
> I just successfully updated my filament list, and to do so I had to copy the new `ota-filament*.zip.sig` file to `/userdata/upgrade/filament/`, remove any other `ota-filament*.zip` files, and [...]

Mine already has that file and the md5 of the file on the printer and the file downloaded are the same. Just checked and for my external spool, I can set it to Bambu Lab PETG HF and ABS-GF, which @Shadow27374 said was missing on his. Not sure what I did to get mine to update automatically…

I don't have any of the new filaments to test if the AMS will read them as the update notes claim that it will.

[image]

[image]

jphannifan, 2024-09-12:
If you can select them from the touchscreen, then you're fully updated and should have no trouble with the RFID features with any of the new filaments. 

I have a roll of ASA-Aero on its way to me, so I can verify this in a couple of days. 

I'm still a little unclear what conditions prevent the printer from downloading the filament file and updating the list automatically, but I think we're pretty close to a solution. Thanks for your help

lightmaster, 2024-11-04:
@jphannifan I got an AMS Hub and it's on the 00.00.00.42. Can't get it to update to 00.00.00.43. I've tried both the version.json (shows up as an update, but doesn't do anything when clicked, and can't get it to update using the mqtt command sent to `device/<printer serial>/upgrade`:

```
{
  "upgrade": {
    "command": "start",
    "sequence_id": "0",
    "module": "ahb",
    "version": "00.00.00.43",
    "url": "http://127.0.0.1/ahb_rev1-firmware-v00.00.00.43-20240511112057.bin"
  }
}
```

or 

```
{
  "upgrade": {
    "command": "start",
    "sequence_id": "0",
    "module": "ahb",
    "version": "00.00.00.43",
    "url": "file:///sdcard/ahb_rev1-firmware-v00.00.00.43-20240511112057.bin"
  }
}
```

malventano, 2024-11-14:
> I just successfully updated my filament list, and to do so I had to copy the new `ota-filament*.zip.sig` file to `/userdata/upgrade/filament/`, remove any other `ota-filament*.zip` files, and reboot.
> ```
> curl  https://public-cdn.bblmw.com/upgrade/device/BL-P001/filament/product/ota-filament-v01.09.00.22-20240902153732.zip.sig -o /tmp/tmp.zip.sig
> mv /tmp/tmp.zip.sig /userdata/upgrade/filament/ota-filament-v01.09.00.22-20240902153732.zip.sig
> ```
> 
> Then try restarting the screen service and/or rebooting to get the list to refresh.

How does one figure out if a more recent list is available / determine the url to pull more recent lists manually?
Bambu just released TPU for AMS, and I'd imagine that requires a newer list.

jphoke, 2025-01-18:
I follow the if it works dont mess with it... and finally needed to figure out getting PETG-HF in the on screen selection...

the above
`curl  https://public-cdn.bblmw.com/upgrade/device/BL-P001/filament/product/ota-filament-v01.09.00.22-20240902153732.zip.sig -o /tmp/tmp.zip.sig
mv /tmp/tmp.zip.sig /userdata/upgrade/filament/ota-filament-v01.09.00.22-20240902153732.zip.sig`

Worked *perfectly* thank you

patschi, 2025-02-15:
I have also been trying to update my AMS firmware, so wanted to share my notes if useful to anyone. I didn't want to mess around with MQTT clients or so, so I discovered a native script which takes care about everything.

**To get the latest firmware updates for AMS:**
https://bambulab.com/en/support/firmware-download/x1

**1. Prepare folders:**
```
[root@BL-P001-sdcard:~]# mkdir -p /mnt/sdcard/update/
[root@BL-P001-sdcard:~]# cd /mnt/sdcard/update/
```

**2. Download AMS firmware (rev8):**
```
[root@BL-P001-sdcard:/mnt/sdcard/update]# wget https://public-cdn.bblmw.com/upgrade/device/BL-A001/00.00.06.44/product/8a38ace450/ams_rev8-firmware-v00.00.06.44-20240703093110_product.bin.sig
Connecting to public-cdn.bblmw.com (172.64.144.14:443)
ams_rev8-firmware-v0 100% |******************************************************************************************************************************************************************************************************************************************************************| 91316   0:00:00 ETA
```

**3. Check file and its checksum:**
```
[root@BL-P001-sdcard:/mnt/sdcard/update]# ls -lsh /mnt/sdcard/update/ams_rev8-firmware-v00.00.06.44-20240703093110_product.bin.sig
    96 -rw-r--r--    1 bblp     sftp       89.2K Feb 15 01:40 ams_rev8-firmware-v00.00.06.44-20240703093110_product.bin.sig
[root@BL-P001-sdcard:/mnt/sdcard/update]# echo "9be1ccccd3ee5e0e63ef9214b322ad5e  ams_rev8-firmware-v00.00.06.44-20240703093110_product.bin.sig" | md5sum -c -
ams_rev8-firmware-v00.00.06.44-20240703093110_product.bin.sig: OK
```

**4. Load MQTT script:**
```
[root@BL-P001-sdcard:/mnt/sdcard/update]# source /usr/bin/mqtt_access.sh
bbl_mpstate[3394]: BSAL:bbl_sal_get_flag:44: read flag type 2 state 1 success.
bbl_mpstate[3394]: BSAL:bbl_sal_get_mpstate:95: read mp_state 2 success.
bbl_mpstate[3395]: BSAL:bbl_sal_get_flag:44: read flag type 0 state 1 success.
bbl_mpstate[3395]: BSAL:bbl_sal_get_sn:168: read sn <SERIAL> len 16 type 1 sucess.
```

**5. Send MQTT to update:**
```
[root@BL-P001-sdcard:/mnt/sdcard/update]# mqtt_pub '{"upgrade":{"command":"start","sequence_id":"0","module":"ams","version":"00.00.06.44","url":"http://127.0.0.1/update/ams_rev8-firmware-v00.00.06.44-20240703093110_product.bin.sig"}}'
Client (null) sending CONNECT
Client (null) received CONNACK (0)
Client (null) sending PUBLISH (d0, q0, r0, m1, 'device/<SERIAL>/request', ... (175 bytes))
Client (null) sending DISCONNECT
```

**Log: /var/log/syslog.log**
```
Feb 15 03:11:31 notice mosquitto[1379]: 2025-02-15T03:11:31: New connection from 127.0.0.1:51348 on port 8883.
Feb 15 03:11:31 notice mosquitto[1379]: 2025-02-15T03:11:31: New client connected from 127.0.0.1:51348 as auto-163EF0FB-5C02-436B-4490-C028CB2ED332 (p2, c1, k60, u'bblp').
Feb 15 03:11:31 info device_gate[1435]: [messageArrived][813] node_name:broker, MQTT MsgArrived Topic: device/<SERIAL>/request TopicLen: 30
Feb 15 03:11:31 info device_gate[1435]: [device_request_callback][54] Find handle for event: upgrade
Feb 15 03:11:31 info device_gate[1435]: [device_request_callback][59] Find forward route for event: upgrade
Feb 15 03:11:31 info device_gate[1435]: [upgrade_event_handler][281] request for command: "start"
Feb 15 03:11:31 notice mosquitto[1379]: 2025-02-15T03:11:31: Client auto-163EF0FB-5C02-436B-4490-C028CB2ED332 disconnected.
Feb 15 03:11:31 info device_gate[1435]: [topic_forward][28] Forward topic succeed:: Topic: device/request/upgrade
Feb 15 03:11:31 info upgrade: [upgrade][json_parse][34] got pub message :{"command":"start","module":"ams","sequence_id":"0","url":"http://127.0.0.1/update/ams_rev8-firmware-v00.00.06.44-20240703093110_product.bin.sig","version":"00.00.06.44"}
[...]
Feb 15 03:11:31 err upgrade: [upgrade][message][248] flashing ams0
Feb 15 03:11:31 err upgrade: [upgrade][message][248] verifying: ams
Feb 15 03:11:31 info upgrade: [upgrade][do_upgrade_mcu][220] init_fw_handler
Feb 15 03:11:31 info upgrade: [upgrade][init_fw_handler][68] mcu upgrade encrypt file
Feb 15 03:11:31 info upgrade: [upgrade][do_upgrade_mcu][229] do_mcu_upgrade_request
[...]
Feb 15 03:11:32 debug bbl_screen: Updater Updater::FLASH_START 0.06
[...]
Feb 15 03:11:58 debug bbl_screen: Updater success msg: "The firmware has been updated to version 99.00.00.00."
[...]
Feb 15 03:11:58 info upgrade: [upgrade_iot][load_device_info_json][113] {"ams":[{"address":0,"dev_model_name":"ams","device_id":"<SERIAL>","hw_version":"AMS08","sw_version":"00.00.06.44","type":"product","version":"00.00.06.44"}][...]
```

Should also work the same way for AMS rev7 or AMS Hub.

Roulendo, 2025-02-18:
@Patschi, thanks a lot for your detailed instructions! I followed your steps carefully, and everything worked exactly as you described. However, at the end, I get a message in the printer window saying:

"Downloading..." followed by "Download failed (0x7000209). Please restart the printer and try updating again."

Any idea what might be causing this? I'm running in LAN-only mode, could that be the issue?

patschi, 2025-02-18:
@Roulendo Can you try `/mnt/sdcard/` instead of `/mnt/sdcard/update/`. I tried both and second worked for me (during second attempt, but maybe I messed it up and only the first did work)

Roulendo, 2025-02-18:
**@patschi**, thanks a lot for your support!  

I just tried the tip from **@jphannifan**, and it worked – the  PETG HF from Bambulab is now recognized in the AMS. 🚀  

Here are the commands I used:  

```bash
curl -o /tmp/tmp.zip.sig https://public-cdn.bblmw.com/upgrade/device/BL-P001/filament/product/ota-filament-v01.09.00.22-20240902153732.zip.sig  
mv /tmp/tmp.zip.sig /userdata/upgrade/filament/ota-filament-v01.09.00.22-20240902153732.zip.sig

Hope this helps someone else! 😊

MrNikFlash, 2025-04-01:
Good day!
I updated the AMS1 firmware according to the instructions MQTT to update, everything worked.
I can't update the AMS2 firmware, when I try, there is an error (0x0000 0105). How do I update the AMS2 firmware and the HUB?

MrNikFlash, 2025-04-03:
Good day!

`mqtt_pub '{"upgrade":{"command":"start","sequence_id":"0","module":"ams/1","version":"00.00.06.44","url":"http://127.0.0.1/update/ams_rev8-firmware-v00.00.06.44-20240703093110_product.bin.sig"}}'`

Doesn't work, causes an error (0x0000 0105). What's wrong with the syntax for update AMS#2 ???

lightmaster, 2025-04-08:
@MrNikFlash I couldn't figure out how to get it to update AMS2, so I reset the AMSes (button titled Reset AMS Information) and only connected AMS2 so it was discovered as AMS1 and was able to be updated.